mikaelpatel/Cosa #290: Add support for Wicked Device WildFire

mikaelpatel, 2015-01-01:


mikaelpatel, 2015-01-01:
See commit https://github.com/mikaelpatel/Cosa/commit/de4a3e5c7f8dae87fa4fc6374aefcb1a7ea0b5c1 for initial support. 

Board extra hardware device are not yet supported (i.e. JTAG, CC3000).

jediunix, 2015-01-01:
FYI I have a CC3000 and when looking into writing a driver I came across some unsettling info  - apparently they are flaky and being abandoned by many.  The Spark Core has it too.  So I got an ESP8266 and, well, its very rough.

mikaelpatel, 2015-01-01:
Do you have any links? I guess a lot has to do with poor power supply. TI devices are often rock solid. In any case I think the CC3000 seems like a nice challenge for a new device driver. And the WildFire board has a lot of SRAM (16 K) so this gives some resources to allow additional multi-tasking support, color off-screen canvas and shadow bitmaps for LCDs. 

The ESP8266 seems also to be an interesting challenge. 

Cheers!

jediunix, 2015-01-01:
https://github.com/cmagagna/ArduinoCC3000 is the one I remember

ESP - I'd say strange.  Very limited control available; error messages are goofy.  Me thinks they put the hardware out long before the software was ready...  Even hardware is very rough - **12** different incarnations.  The ESP-05 apparently can't have its firmware upgraded because they didn't expose the programming pin - and the "cloud" update bricked mine :(

mikaelpatel, 2015-01-02:
Thanks for the warning. After reading about the problems on the TI forum and some of the other comments above I think this product (CC3000) is very immature. And actually already replaced by TI.

In any case I will continue to have a look at the SPI command protocol and how that can be used to implement the Socket/Wireless interfaces. 

Cheers!

mikaelpatel, 2015-01-15:
So far the following has been implemented for the WildFire.

1. Board definition with SPI chip select pins initialized to correct state (Board::init). https://github.com/mikaelpatel/Cosa/commit/771379a46ee5cd50d758d83369c4e414ea420893
2. Cosa RTC.
3. Cosa Watchdog.
4. UART.
5. SPI.
6. TWI.
7. SD. https://github.com/mikaelpatel/Cosa/commit/9595aa3d5cecdaec274686e72dfbcbe73342c43a
8. Flash W25X40CL. https://github.com/mikaelpatel/Cosa/commit/43c63f33efd8d5993a5f40eac5a3be471d578ce3

mikaelpatel, 2015-01-20:
For file system on Wildfire SPI flash memory please see issue https://github.com/mikaelpatel/Cosa/issues/213

mikaelpatel, 2015-01-20:
The backlog is now:
1. RFM69.
2. CC3000.
3. WildFire Watchdog.

closed 2015-05-15 by mikaelpatel