mikaelpatel/Cosa #291: ST7735 broken?

jediunix, 2015-01-11:
CosaCanvasFonts.ino is producing strange output.  Left 2/3 of screen is a mostly black sorta grid.  The right 1/3 has font characters overlaying each other.

Both my SPI & TWI OLED are working for the same font.

CosaCanvasDemo is broken too...

I see you just made some changes there so I will leave the debugging to you :)

mikaelpatel, 2015-01-11:
The obvious question is if you are on the latest updates? I corrected that problem a few days ago.

mikaelpatel, 2015-01-11:
You need to configure CosaCanvasFonts.ino for ST7735. The default is right now ILI9341. That will give a lot of problems run on a ST7735. See https://github.com/mikaelpatel/Cosa/blob/master/examples/Canvas/CosaCanvasFonts/CosaCanvasFonts.ino#L87

mikaelpatel, 2015-01-12:
All Canvas example sketches are now back to default ST7735.

jediunix, 2015-01-12:
Oops.  I didn't see that line.  I was thrown by the fact that I could see something :)

closed 2015-01-12 by jediunix

mikaelpatel, 2015-01-12:
No problems. Actually this forced me to take a decision on which driver should be default. BW: Have you checked the performance of the new driver structure. Added some extra tricks and increased the SPI clock frequency.