mikaelpatel/Cosa #293: Font question - unprintable characters

jediunix, 2015-01-12:
I noticed that `Textbox::putchar` only prints characters >= ' '.  Yet `System5x7` has < ' ' (except tiny).

The fonts I added all have 0x0-0x7f and some of them *could* have all 256.

Is ' ' through '~' the range you want to support?

If you want to support "extended" ASCII (>128) then `char` types through text handling should be changed to `unsigned char`.  Maybe makes sense to create a global type `char_t`.

I don't really have an opinion either way.  I can see value to supporting a full character set yet space is of concern in this environment.  Then there are the "special" characters '\n', '\r', '\f'.

I'm working on simple compression at the moment.  Another thought is that all fonts could have entire 256 set and any characters not present would not add any overhead (besides the smaller fonts growing due to the lookup table).  Maybe in this scenario it would be best to leave `System5x7` uncompressed and never having characters below ' ' or above '~'.

Finally, how should a character that is not in a font be presented?  As a blank or some kind of graphic to alert the viewer?

mikaelpatel, 2015-01-12:
Hum, seems like a thing that has not got updated. The Textbox should check that the character is available in the font otherwise skip or mark the character as missing. The second part is difficult as it assumes that there is a special character available for this. Guess that a block of OxFF could do the trick, i.e., the Font::get_bitmap should answer with a pointer to the block for bad characters??

Cheers!

jediunix, 2015-01-12:
My current working code returns 0x55 for every byte requested for a character that doesn't exist.  This gives a bounding box filled with horizontal lines every other row.  Not really a need to take up font space with a special character - also means if somebody added a private font they wouldn't need one.

I'll change Textbox to `font->draw()` for every character other than the special ones - leave it up to draw to decide what to do.

closed 2015-01-13 by jediunix