mikaelpatel/Cosa #297: Suggest "Text" could use a refactoring

jediunix, 2015-01-17:
I enhanced my CosaOLED drivers to support "raw" text mode (no interpretation of control characters so font characters for them can be displayed) and while I was at it I added support for underline mode.
https://github.com/jediunix/CosaOLED/blob/master/OLED_Text.hh#L37-L43

I considered submitting a PR to enhance other drivers to do the same.  I came to the conclusion that text handling could use a refactoring.  There should be a common way to interact with displaying text.  `Textbox` is derived from `IOStream::Device` as is `LCD`.  However, `LCD::Device` adds things like `display_clear()`, `get_cursor()`, `set_cursor()`, `get_tab_stop()`, `set_tab_stop()`, and `set_text_mode()` (and some drivers have `line_clear()`).  None of these are specific to a *device* and all of them should be available in `Textbox`.

Thanks

mikaelpatel, 2015-01-18:
Hum, this seems to be more than one issue. 

First, should small text LCD devices support special characters? This seems to do with how much an application should be concerned with layout. In the best of worlds we would have something like a tiny version of HTML output and a screen rendering process. But with the limited resources this is not realistic. In some sense the special characters are "short forms" for display_clear() and set_cursor(). Backspace is a short form for set_cursor(); putchar(' '); set_cursor(). Anyway I understand your concern especially when using "new line/carriage return". Most small LCD screens don't really need "scrolling". And if they should do scrolling some type of output buffering is needed to make "scrolling" more realistic. It is of ST7565 that does something that looks like scrolling. The rest of the LCD drivers are mock-ups when it comes to scrolling. 

The second issue, as I understand it, is refactoring to a common interface for text output/input; Screen/Console. And as you point out this ripples on a few classes; LCD and Canvas::Textbox. Obvious this would be an interesting enhancement as reconfiguration of an application to an other display becomes so much easier.

Hope I understood this/these issues you are suggesting. The next step is to get this a bit more detailed with some kind of list of changes and ripple effects. In any case I fully agree on that the Canvas::Textbox needs more functionality. Especially cursor control. 

Cheers!

jediunix, 2015-01-18:
Yep, many issues here.  I have not thought about this much.  The device types are intermixed with this conversation so I apologize ahead of time for expanding this issue further.

I guess the first thing to consider is what does *small text LCD devices* mean? My OLED devices, SSD1306/SH1106, are 128 pixels wide and 64 high.  With 5x7 that's 21 characters and 8 lines.  Larger monochrome OLEDs exist.

For displaying text I would suggest the goal should be a subset of what a "terminal" can do.  Cursor positioning, erasing display/line, inverse/underline, raw, vertical scrolling at a minimum (perhaps controlled by a text mode).  Scrolling should be independent of the device - use hardware scrolling when available and buffered redraw when not.  Even for 2 line LCD.

At what size does a textbox make sense?  Wouldn't that be the way to mix fonts on a display?

On first glance I thought the monochrome OLED would be a Canvas device however there are issues.  First they are monochrome and Canvas looks to be color only (I suppose one could kludge a mapping in the driver for colors).  A bigger issue is that Canvas devices are all pixel addressable - the monochrome OLEDs are not.  So to do Canvas would require a full framebuffer which is too expensive in most cases (which then means to fully support them there has to be 2 implementations, one as LCD and another as Canvas).  I suppose it is possible that there are pixel-addressable monochrome controllers out there.

I don't have the next level of detail to propose as there are several potential paths here.  It seems to me there are 3 types of display devices - barebones not Font-capable, Font-capable monochrome and not pixel-addressable, and Font-capable color/grayscale pixel-addressable.  Color/gray/pixel-addressable as a subclass of Mono/non-pixel-addressable?  My gut feel is that Font-capable device drivers should know nothing about Fonts or Textboxes.

Feel free to replace this issue with multiple new ones based on how you think this should proceed...

PS These tiny monochrome OLED displays are desirable because they are very small/dense, bright, energy efficient, and cheap.

closed 2015-02-08 by mikaelpatel