mikaelpatel/Cosa #305: TWI::Slave does not seem to work.

oddstr13, 2015-01-25:
I'm having trouble getting the TWI slave example to work with either the TWI scan sketch, or i2cdetect on the RPi.
I'm not quite sure if this is an actual bug, or if I am doing something wrong.

* The BMP085 shows up on the I2C bus, both for the scan sketch and for the Pi.
* Once the TWI::Slave slave is connected, i2cdetect takes about 0.7s, and no devices are detected.
* A scan with i2cdetect without the Cosa TWI::Slave connected takes about 0.1s, and detects 0x77, which is the BMP085.
* Reversing the SCL and SDA pins on the TWI::Slave causes i2cdetect to detect devices on all addresses. This would be expected, I guess?

Setups tried
* Arduino UNO with the plain CosaTWIslave example sketch.
* ATtiny85 with my modified sketch (NEXA::Transmitter control from the Pi is the goal; http://oddstr13.openshell.no/paste/VMSYQL7T/)  the LED toggle was added so I could see if there was actually anything running.
* Breadboarded Arduino UNO (ATmega328P with 16MHz crystall & Optiboot) with my NEXA transmitter sketch. Also tried with the CosaTWIslave example sketch.

Connecting the SCL pin of the Cosa TWI::Slave is what triggers this issue.
* SDA connected, SCL _NOT_ connected == Pi can access BMP085.
* SDA connected, SCL connected == Pi can _NOT_ access the BMP085.
* SDA _NOT_ connected, SCL connected == Pi can _NOT_ access the BMP085.

Having only the TWI::Slave on the bus results in same results as with the BMP085 connected, except that the BMP085 is not detected then.

Unfortunately I do not have a scope or a logic analyzer, so I'm not able to debug the issue on the electrical level.

* Cosa version is commit 9350f31c14ba8f1c754ad4dece69beedc55d8a26
* Arduino version is 1.0.6

I can update to latest commit and try again, but I did not see any commit messages that suggested TWI might have been touched.

PS:
phew, that ended up as quite the wall of text..
Is there an IRC channel for Cosa?

mikaelpatel, 2015-01-25:
It is over six months since I rigged up the TWI master-slave circuit. Cosa has developed a lot since. I would recommend that you start with the example sketches for TWI master/slave. And if that does not work there is an issue with possible changes. All other changes and/or integration that you have done is basically asking for development support. I have limited time to provide that - Sorry about that. 

Two important issue; 1) common ground, 2) pull-up resistors. 

Cheers! Mikael

oddstr13, 2015-01-26:
I did have common ground. I did forget the pull-up resistors, however after looking into it, the BMP085 breakout from SparkFun has onboard pull-ups enabled by default, so this shouldn't be an issue as long as it is connected. I did test just the ATtiny again, with the added pull-up resistors with a value of 4.6k Ohm, and the result is the same as last time. The i2cdetect program uses extra long time to scan the I2C bus, returning no results.

The base of my sketch is the CosaNEXAsender  and the CosaTWIremotePinSlave, but that is beside the point, as I am having trouble getting either of the CosaTWI*slave sketches to work on their own.

-----

It was not my intention to give the impression of asking for help with my particular application. I simply wish to locate where the TWI::Slave is failing for me, if it is my fault, or if the cause is a bug in the Cosa code.
I have read other issues on here that mention problems related to using the Arduino IDE, which I am, as that is what was suggested in the [install instructions](https://github.com/mikaelpatel/Cosa/blob/master/doc/02-install.md).
**1)** Are there any instructions available on how to use Cosa outside of the Arduino IDE?

CosaTWISlave on COM7, CosaTWIMaster on COM4. Both are Arduino Uno boards.
Code is compiled and uploaded with Arduino 1.0.6 on Windows 7.

[image]

The result of starting the slave, then the master:
[image]

Result of hitting reset on the slave:
[image]

The result of starting the master, then the slave:
[image]

Then hitting reset on the slave again:
[image]

-----

**2)** I would appreciate if someone other than me could test TWI::Slave, and see if it is working. This would give an indication of weather the error is likely somewhere in my setup or not.

**3)** Does the Cosa project have an IRC channel, where I can chat with other users of Cosa? Official channel or otherwise.

mikaelpatel, 2015-01-26:
@oddstr13 

That was the best issue report ever. Thanks for waking the extra mile. I took a few minutes to connect two Arduino Nano and run the TWI Master/Slave example sketches with exactly the same result (no extra external pull-up resistors). Took a look at the code and found the problem more or less directly. 

The major changes on the TWI has been to support multi-tasking. The change that was the root of the problem was an update of the default loop() function. Previously it was the event dispatcher but lately updated to exit(). This is a ripple effect of this change. 

The TWI Slave requires an event dispatcher. That what was missing. See commit https://github.com/mikaelpatel/Cosa/commit/05e9ec485743e8027b1f18f7f3212a91b3d36841.

There is no IRC channel for Cosa. So far I have used the Arduino forum to post notices on update and this issue list for bugs and enhancement requests. There is a blog but it is starting to be very out of date. 

It is possible to use Cosa without the Arduino IDE using the command line support. Please see INSTALL.txt and the post on the Arduino forum. This is currently only supported on Linux.

Cheers! Mikael

oddstr13, 2015-01-26:
@mikaelpatel Excellent! The test setup now works, and the Pi can see the Uno slave on the I2C bus.
CosaTWIremotePinSlave also needs an update, and I have created a pull-request for that.

closed 2015-01-26 by mikaelpatel

mikaelpatel, 2015-02-14:
Correcting labels as this is not a bug on TWI::Slave but on the demo CosaTWIslave example sketch. There was a missing event dispatcher. No changes to the actual TWI code.

dothebart, 2018-03-06:
sorry for being a bit late to the party - is any of the slave examples expected to show up when running i2cdetect? 
Using an [seeduino lotus](http://wiki.seeed.cc/Seeeduino_Lotus/) I can see it output on the tty on the usb interface: 

```
CosaTWISlave: started
78:void setup():trace:free_memory() = 1638
79:void setup():trace:sizeof(Echo) = 14
80:void setup():trace:sizeof(OutputPin) = 4
```
I'm connecting it on the `I2C..` which seems to be connecting the same ports of the chip as `I2c.`

Or do I have to connect one of the analog ports?

mikaelpatel, 2018-03-06:
@dothebart It is awhile since I worked on this :) but I will try to help you out. How have you connected, wired, the devices? Pullup resistors? Gound? The I2C pins are A4 and A5 on a standard ATmega328X based Arduino so that should not be an issue. An I2C scanner should detect the slave if it acknowledges the address. This part is done by hardware so this is an indication of a wring issue. Cheers!

dothebart, 2018-03-07:
[the schematics say there are pullups onboard](https://github.com/SeeedDocument/Seeeduino_Lotus/raw/master/res/Seeeduino%20Lotus%20v1.1%20SCH.pdf) and I have been using [a modified grovepi firmware](https://github.com/dothebart/GrovePi/blob/master/Firmware/Source/v1.2/grove_pi_v1_2_7/grove_pi_v1_2_7.ino) which has been working - sort of - frequent disconnects etc. which was why I'm looking at Cosa. However, meanwhile I've switched from a PI1 to a Bananapi - maybe I need to re-test with the grovepi firmware whether its supposed to be working...

dothebart, 2018-03-11:
Ok, I've re-validated with the grovepi sketch and a rigol DS1054 that it actually does something when talking to the banana pi, but the cosa examples don't work - silence in i2detect.

mikaelpatel, 2018-03-11:
@dothebart Do you have a logic analyzer? I guess that the banana pi is a 3.3V system, and you are interfacing a 5V ATmega328X. I would try to verify the hardware wiring first. Ground and logic levels is a source of hardware issues. You might need a logic level converter? Cheers!

dothebart, 2018-03-11:
ok, on the Rigol one sees the [master sketch](https://github.com/mikaelpatel/Cosa/blob/master/examples/TWI/CosaTWImaster/CosaTWImaster.ino) sending read commands  as mad. I've tried that on two Lotus' connected on their I2c Bus'es. I can see the reads on `I2C.` and `I2c..` of Lotus A, and when connected via `I2C..` on the `I2C.` of the other. 
So both I2C parts of the two chips work, they should be able to talk to each other. 
I can see the slaves prints on the serial console during startup, then its silent. 
On the master I've added some more prints, and I see it looping and waiting for a reply from the slave. 
Is that how its supposed to work?

dothebart, 2018-03-11:
Yes, [the bananapi is connected via a logic converter](https://www.exp-tech.de/en/modules/currentvoltage/5410/sparkfun-logic-level-converter-bi-directional) On the Oscar you see that the signal is not as clean as on the 3.3V side, but with a genuine arduino wire-lib as used by the grove-pi i2cdetect can find the grove-pi, but not the [cosa slave example](https://github.com/mikaelpatel/Cosa/blob/master/examples/TWI/CosaTWIslave/CosaTWIslave.ino)

dothebart, 2018-03-11:
I can see prints from before the slave entering the loop(), but no prints from `on_request()` or `Echo()`
I can see the master loope endless in: 
```
  time = RTT::micros();
  twi.acquire(&dev);
  do {
    count = twi.read(buf, sizeof(buf));
    trace << PSTR("read while :") << time << ':' << time/(sizeof(buf) - 4) << ':' << count << '\n';
  } while (count < 0);
```
(which one also sees on the Rigol with an ever stopping chain of read commands)

dothebart, 2018-03-11:
nb, regarding the wiring - since with a stock Arduino `wire.h` based grovepi firmware i2cdetect on the banapi can see the Lotus, I don't think there is an issue on the hardware side, the software is the problem here. The logic analyzer also showed that the connection between the two Lotus'es also works, the master example also does what it should. Is this supposed to work with two similar systems?

dothebart, 2018-03-11:
Ok, I've downgraded my cosa environment to 05e9ec485743e8027b1f18f7f3212a91b3d36841 - the version that was the previous last result of this ticket. Now the loop on the master actually does usefull things: 

```
Terminal ready
CosaTWImaster: started
50:void setup():trace:free_memory() = 1635
51:void setup():trace:sizeof(OutputPin) = 4
52:void setup():trace:sizeof(TWI) = 34
write(2):348:174:000376: 000 000 
read while :41020:10255:8
                         read(8):28756:3594:002271: 001 002 003 004 005 006 007 008 
read while :139560:34890:4read(4):29748:7437:002271: 001 002 003 004 

write(2):348:174:000376: 001 000 
read while :2226700:556675:8
                            read(8):33172:4146:002271: 002 003 004 005 006 007 008 009 
read while :2329660:582415:4read(4):32032:8008:002271: 002 003 004 005 

write(2):348:174:000376: 002 000 
read while :4418692:1104673:8
                             read(8):34328:4291:002271: 003 004 005 006 007 008 009 010 
read while :4522796:1130699:4read(4):33180:8295:002271: 003 004 005 006 

write(2):348:174:000376: 003 000 
```
so I guess there is a regression in Cosa regarding the slave example?

dothebart, 2018-03-12:
ok, it broke between 519c110a85457a6fc68de845888e946c51549894 and 1ed39561b93937cd48995352f101d26230ac4f53 - going to narrow it tonite .

dothebart, 2018-03-12:
git bisect bad 
173c3ebaa483d09acefb62df920bc6b064bf4420 is the first bad commit
commit 173c3ebaa483d09acefb62df920bc6b064bf4420
Author: Mikael Patel <mikael.patel@gmail.com>
Date:   Sat Aug 15 00:00:58 2015 +0200

    Refactoring TWI::Driver/Slave classes; improving support for asychronous calls.

:040000 040000 82f82d95f2311e3c40385ef4825da7f5cdf26252 e0cfd51c1a2cc5d270981d1052586512bae8a965 M	cores

mikaelpatel, 2018-03-12:
@dothebart I have seen that you have been active on this issue. Unfortunately I am on holiday and have limited access to both test equipment and development environment. Difficult to give any hints right now.

dothebart, 2018-03-12:
if ` git show 173c3ebaa483d09acefb62df920bc6b064bf4420` rings a bell with limited bandwith, or I should try anything let me know. else enjoy your vacation.

dothebart, 2018-03-13:
hm, revoking the multiple inheritance in the slave class and moving that back to the Driver doesn't fix it.

dothebart, 2018-03-13:
Ok, with the downgraded version it also shows up: ```
 i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- 5a -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  ```

dothebart, 2018-03-26:
Hm, I've tried to get async analog reading to work, however, it fails similarily: https://gist.github.com/dothebart/18a14b75a3af822d0473b4dd8fc8661a
Looking through the drivers and their examples also didn't get me something working :-( 
Any hints on what I'm doing wrong?

mikaelpatel, 2018-03-26:
The simplest solution is to use the sample_request() and sample_await() member functions. Adding Job, Periodic, Scheduler will require a state-machine. Also the state-machine needs to be coupled to the AnalogPin interrupt and event handler. This requires a fair amount of understanding of the mechanisms. I would recommend starting with the pure async-calls; sample_request() etc.

dothebart, 2018-03-26:
You're referencing `examples/Tools/CosaProbe/CosaProbe.ino`? 
If I get that correctly it still will wait for analogpins to deliver their value? 

While I've found lots of your nice (working) basic examples, The situation doesn't seem to be as good in the areas that set apart cosa over the standard arduino libs; did you have the chance to have a look at whats wrong with the TWI-Slave example as of https://github.com/mikaelpatel/Cosa/commit/173c3ebaa483d09acefb62df920bc6b064bf4420 ? Would shell access to an arduino with two boards connected help you in finding a fix?

mikaelpatel, 2018-03-27:
No I am not referencing that example. The API design is somewhat the same but what I was referencing was the AnalogPin API. It allows asynchron calls. 

But this is only one of the issues you are actually asking about. The second issue is how to do this periodically. As you have already seen there are many supported levels in Cosa. It all depends on the frequency and latency. The fastest is the busy wait variant. Often called await(). The next level is RTT Scheduler and Job with interrupt level calls. Events is the next level and has the highest latency as the handler is called much later from the Event loop. 

Unfortunately I am busy working on a consultant assignment and will not have time to look into the TWI-Slave issue for some time.