mikaelpatel/Cosa #308: SNMP MIB OIDs have a mangled root due to missing encoding

dansut, 2015-01-27:
The first byte of a ASN.1 BER encoded OID for SNMP should always be 0x2B as it has strangely compressed form of 1.3 => (40*1)+3 = 43 = 0x2B.

As is this is not done in the current Cosa SNMP implementation the effect is that of adding a 0. to the beginning of every OID and misplacing things in the 0.1.3.6.1 root as opposed to 1.3.6.1 where they really belong.

mikaelpatel, 2015-01-28:
Updated the OIDs with the correct prefix. Have not test it on a board with W5100 yet. Please see commit https://github.com/mikaelpatel/Cosa/commit/c60eb1b56d0f4db0560f65fe7614bb43ee53f9e7.

dansut, 2015-01-29:
Thanks for fix, I don't know why I didn't think to just do that.

Tested, and it works for snmpget on mib-2 stuff but CosaSNMP.ino example needs fixing for it to work with Arduino MIB - will give you a pull request on that.

However snmpwalk is broken by this change. I don't think it was ever supposed to work how documented in the example comment though ... leaving MIB off walk should start from 1.3.6.1.2.1 which is reasonable behavior.

mikaelpatel, 2015-01-29:
Thanks for testing this. Much appreciated. I assumed there was going to be a few ripple effects on the examples, etc.

dansut, 2015-01-30:
I think I have made snmpwalk again, and improved the example a little see pull request #313 
Cheers

closed 2015-02-08 by mikaelpatel