mikaelpatel/Cosa #315: CosaTWIslave does not work on attiny85 or attiny84

oddstr13, 2015-01-31:
CosaTWIslave does __not__ show up on the I2C bus when uploaded to:
 * attiny85-8
 * attiny84-8

CosaTWIslave __does__ show up when uploaded to:
 * uno
 * atmega328-8
 * mega2560

My current definition of _show up_ is the following python script ran on a Raspberry Pi B1, outputting `0x5a` as one of the available devices on the bus.
```python
import smbus
bus = smbus.SMBus(0)

def scan(start=0x00, stop=0x7f):
    res = []
    for addr in xrange(start, stop+1):
        try:
            bus.read_byte(addr)
            res.append(addr)
        except IOError:
            pass
    return res

print(["{0:02x}".format(addr) for addr in scan()])
```

mikaelpatel, 2015-01-31:
Attiny must have pullup and double check the pins.

oddstr13, 2015-02-03:
Pull-up is present.
Pins are double checked, triple checked and so on. They are the same as MISO and SCK SPI pins on the attiny84 and attiny85.

One of these days, I'll get myself a saleae, so that I can see what's going on. There are a few tools I would like to get before it however, like a proper soldering station.

I see that the attinys use a different backend for the I2C, might be related. [cores/cosa/Cosa/TWI.hh#L25](https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/TWI.hh#L25)

mikaelpatel, 2015-02-08:
Yes the attinys use USI. They do not have true TWI support. The TWI protocol is "manually" parsed.

oddstr13, 2015-02-10:
Possibly related to the issue:
http://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html
Would need a logic analyzer or oscilloscope to confirm.

mikaelpatel, 2015-02-10:
@oddstr13 

Thanks for continuing the research on this issue. This gives me a good lead on what could be done. There are some timing constants in the Cosa TWI implementation for USI (ATtiny) that you can tweak. https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/USI/USI_TWI.cpp#L25

You could start by turning on USE_FAST_MODE.

After that we would need to know what is the timing to keep within the non-stretch limit.

Cheers!

mikaelpatel, 2015-02-10:
@oddstr13 

Could you please confirm that the ATtiny (TWI slave) shows up when scanned from an Arduino?

oddstr13, 2015-02-10:
The CosaTWIscanner sketch does indeed see the ATtiny84/85.

They do seem to freeze up from time to time, not quite sure why, but it seems to be related to me moving around the wires for the i2c bus on my breadboard. Resetting tiny fixes it. Likely a separate issue.

mikaelpatel, 2015-02-10:
Thanks for the update even if it actually puts an issue on the table that I have avoided; TWI timeout. Currently there is no error state when the protocol fails. Robustness could be improved if the TWI Slave is to have a broader usage.

closed 2015-02-14 by mikaelpatel