mikaelpatel/Cosa #316: `cosa leonardo upload` fails with compile error on the file PWMPin.cpp

oddstr13, 2015-01-31:
Possibly related to the old issue #229

```cpp
oddstr13@oddlaptop-Arch:~/lib/Cosa/examples/Blink/CosaBlink
% ISP_PORT=/dev/ttyACM1 MONITOR_PORT=$ISP_PORT cosa leonardo upload                                                                                                             [0]
mkdir -p /home/oddstr13/lib/Cosa/obj/build-1.5.8/leonardo
/usr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c   -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=100 -I. -I/home/oddstr13/lib/Cosa/cores/cosa -I/home/oddstr13/lib/Cosa/variants/arduino/leonardo   -Wall -ffunction-sections -fdata-sections -g -Os -DUSB_VID=0x2341 -DUSB_PID=0x8036 -fno-exceptions  -Wextra -std=gnu++0x -felide-constructors CosaBlink.ino -o /home/oddstr13/lib/Cosa/obj/build-1.5.8/leonardo/CosaBlink.o
/usr/bin/avr-g++ -MMD -c -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=100 -I. -I/home/oddstr13/lib/Cosa/cores/cosa -I/home/oddstr13/lib/Cosa/variants/arduino/leonardo   -Wall -ffunction-sections -fdata-sections -g -Os -DUSB_VID=0x2341 -DUSB_PID=0x8036 -fno-exceptions  -Wextra -std=gnu++0x -felide-constructors /home/oddstr13/lib/Cosa/cores/cosa/Cosa/PWMPin.cpp -o /home/oddstr13/lib/Cosa/obj/build-1.5.8/leonardo/core/Cosa/PWMPin.o
In file included from /home/oddstr13/lib/Cosa/cores/cosa/Cosa/Pin.hh:25:0,
                 from /home/oddstr13/lib/Cosa/cores/cosa/Cosa/OutputPin.hh:24,
                 from /home/oddstr13/lib/Cosa/cores/cosa/Cosa/PWMPin.hh:24,
                 from /home/oddstr13/lib/Cosa/cores/cosa/Cosa/PWMPin.cpp:21:
/home/oddstr13/lib/Cosa/cores/cosa/Cosa/PWMPin.cpp: In member function 'void PWMPin::set(uint8_t)':
/home/oddstr13/lib/Cosa/cores/cosa/Cosa/PWMPin.cpp:175:21: error: 'COM2A1' was not declared in this scope
     bit_set(TCCR3A, COM2A1);
                     ^
/home/oddstr13/lib/Cosa/cores/cosa/Cosa/Bits.h:27:28: note: in definition of macro 'bit_mask'
 #define bit_mask(b) (1 << (b))
                            ^
/home/oddstr13/lib/Cosa/cores/cosa/Cosa/PWMPin.cpp:175:5: note: in expansion of macro 'bit_set'
     bit_set(TCCR3A, COM2A1);
     ^
/home/oddstr13/lib/Cosa/build/Arduino-Makefile/Arduino.mk:1089: recipe for target '/home/oddstr13/lib/Cosa/obj/build-1.5.8/leonardo/core/Cosa/PWMPin.o' failed
make: *** [/home/oddstr13/lib/Cosa/obj/build-1.5.8/leonardo/core/Cosa/PWMPin.o] Error 1
```

mikaelpatel, 2015-01-31:
See build logs https://github.com/mikaelpatel/Cosa/blob/master/build/arduino-1.5.8.log. Looks like something in your configuration. Check compiler version.

oddstr13, 2015-02-03:
The problem was that `ARDUINO_DIR` was not set. Cosa was using the system avr-g++ at version 4.9.2 instead of the avr-g++ bundled with arduino 1.0.6 at version 4.3.2.

The issue wasn't so obvious, as all boards except the atmega32u4 based ones where compiling fine.

I also did a full system update before I found the issue. That might or might not have changed things.

#### Before setting `ARDUINO_DIR`
```bash
% cosa leonardo config                                                                                                                                                          [0]
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDMK_DIR = /home/oddstr13/lib/Cosa/build/Arduino-Makefile 
- [USER]               ARDUINO_DIR = /home/oddstr13/opt/arduino-1.5.8 
- [DEFAULT]            ARDUINO_VERSION = 100 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/oddstr13/.arduino/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/oddstr13/scripts/arduino/sketchbook (from arduino preferences file)
- [AUTODETECTED]       AVR_TOOLS_DIR = /usr (found in $PATH)
- [USER]               ARDUINO_LIB_PATH = /home/oddstr13/lib/Cosa/libraries 
- [USER]               ARDUINO_CORE_PATH = /home/oddstr13/lib/Cosa/cores/cosa 
- [USER]               ARDUINO_VAR_PATH = /home/oddstr13/lib/Cosa/variants 
- [USER]               BOARDS_TXT = /home/oddstr13/lib/Cosa/boards.txt 
- [DEFAULT]            USER_LIB_PATH = /home/oddstr13/scripts/arduino/sketchbook/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = leonardo 
- [COMPUTED]           OBJDIR = /home/oddstr13/lib/Cosa/obj/build-1.5.8/leonardo (from BOARD_TAG)
```

#### After setting `ARDUINO_DIR`
```bash
% cosa leonardo config                                                                                                                                                          [0]
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDMK_DIR = /home/oddstr13/lib/Cosa/build/Arduino-Makefile 
- [USER]               ARDUINO_DIR = /usr/share/arduino 
- [AUTODETECTED]       ARDUINO_VERSION = 106 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/oddstr13/.arduino/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/oddstr13/scripts/arduino/sketchbook (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /usr/share/arduino/hardware/tools/avr (in Arduino distribution)
- [USER]               ARDUINO_LIB_PATH = /home/oddstr13/lib/Cosa/libraries 
- [USER]               ARDUINO_CORE_PATH = /home/oddstr13/lib/Cosa/cores/cosa 
- [USER]               ARDUINO_VAR_PATH = /home/oddstr13/lib/Cosa/variants 
- [USER]               BOARDS_TXT = /home/oddstr13/lib/Cosa/boards.txt 
- [DEFAULT]            USER_LIB_PATH = /home/oddstr13/scripts/arduino/sketchbook/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = leonardo 
- [COMPUTED]           OBJDIR = /home/oddstr13/lib/Cosa/obj/build-1.5.8/leonardo (from BOARD_TAG)
```

closed 2015-02-03 by oddstr13