mikaelpatel/Cosa #317: SPI Endianness

kc9jud, 2015-02-02:
Hello,

I'm writing an interface for the Microchip MCP4822 DAC (Digital-to-Analog Converter). I'm having trouble transferring commands to the chip because of problems with endianness. The chip takes 16-bit instructions, big-endian. I expected to be able to use this to transfer an instruction:

```c++
void
MCP4822::issue(uint16_t cmd)
{
    spi.acquire(this);
    spi.begin();
    spi.write(&cmd, sizeof(cmd));
    spi.end();
    spi.release();
}
```

The chip doesn't respond to those commands. Instead, I have to do something like this:

```c++
void
MCP4822::issue(uint16_t cmd)
{
    spi.acquire(this);
    spi.begin();
    // switch first and second byte positions
    uint16_t mod_cmd = ((cmd>>8) & 0x00FF) | ((cmd<<8) & 0xFF00);
    spi.write(&mod_cmd, sizeof(cmd));
    spi.end();
    spi.release();
}
```

Is there a "right way" to set the endianness for sending multi-byte streams?

Thanks!

mikaelpatel, 2015-02-02:
There is no clear answer to this as it depends on several factors. The most important is should the code be portable between processors with different endians. So far I have tried to keep things simple and just focus on AVR (little-endian). 

This implies that you can solve this at compile (code write) time by swapping the operation code in the definition of the commands. If you do that the problem goes away. Simply send the command as a byte sequence. The same for multi-byte status code etc. Simply swap the bit-field definitions.

If you still want to do byte swap and be portable you should use hton() and ntoh(). You can find support for this in the Cosa/Types.h file. There is a whole set of byte swap functions. 

Cheers!

mikaelpatel, 2015-02-02:
Aha now I see what you are asking; Could the multiple byte TWI write() member function be extended with an endian parameter? Something like twi.write(buf, size, endian) and the parameter endian would control if the buffer is transmitted in reverse order or not? Default would be host-order (little-endian). 

Why not? I will consider adding that. But still you can solve this at compile time.

Cheers!

mikaelpatel, 2015-02-02:
I looked at the product specification http://ww1.microchip.com/downloads/en/DeviceDoc/22249A.pdf
A possible interface is:
```
MCP4822::DACPin(uint8_t channel, uint8_t gain);
void MCP4822::DACPin::set(uint16_t voltage);
```
The control bits are added as late as possible. The member function set() would do the work something like this:
```
spi.acquire(this);
spi.begin();
spi.write(((voltage >> 8) & 0xf) | m_cntrl);
spi.write(voltage);
spi.end();
spi.release();
```
I am assuming that the member data m_cntrl contains the control bits at the right position. 

Cheers!

closed 2015-02-08 by mikaelpatel

kc9jud, 2015-02-14:
Just to follow up on this, I ended up using this function:

```c++
void
LS7366R::issue(uint16_t cmd)
{
    spi.acquire(this);
    spi.begin();
    // this chip expects instructions in big-endian format
    cmd = swap(cmd);
    spi.write(&cmd, sizeof(cmd));
    spi.end();
    spi.release();
}
```

In case you ever add big-endian processors, it might be useful to have a set of "big_endian" and "small_endian" byte manipulation functions that are selected by the processor at build time:

```c++
// On a little-endian processor, LITTLE_ENDIAN would be defined,
// while on a big-endian processor, BIG_ENDIAN would be defined.
#ifdef BIG_ENDIAN

inline uint16_t big_endian(uint16_t value) __attribute__((always_inline));
inline uint16_t
big_endian(uint16_t value)
{
    return value;
}

inline uint16_t little_endian(uint16_t value) __attribute__((always_inline));
inline uint16_t
little_endian(uint16_t value)
{
    return swap(value);
}

#else

inline uint16_t big_endian(uint16_t value) __attribute__((always_inline));
inline uint16_t
big_endian(uint16_t value)
{
    return swap(value);
}

inline uint16_t little_endian(uint16_t value) __attribute__((always_inline));
inline uint16_t
little_endian(uint16_t value)
{
    return value;
}
#endif
```

kc9jud, 2015-02-14:
I've noticed that many SPI devices send things as big-endian if they are MSB-first, so this might turn into a more common problem as more SPI devices get implemented.