mikaelpatel/Cosa #318: Add support for SSD1306 based OLED (128x64)

mikaelpatel, 2015-02-02:


jeditekunum, 2015-02-02:
Since I actually have all that implemented I would be willing to contribute it.  jeditekunum/CosaOLED

You will also want 128x32 version.  SH1106 is almost exactly the same and is widely available from china very cheaply.

I hadn’t/didn’t develop it to PR into Cosa as there are code organizational issues I didn’t want to deal with at the time.  Like

1. It seemed odd to me that “driver”/TWI/SPI stuff was repeated in things in Cosa.

2. There are many SSD/SH devices that are almost identical.  What is the “right way” to implement them without duplicating lots of stuff.

3. The “Canvas problem” that I’ve brought up before - these devices are not pixel addressable.  Is a huge frame buffer the right thing to do?

It may be that you wish to reimplement yourself.  I don’t really care one way or another.  Perhaps other significant refactoring is a reason for you to do it yourself.  This part is why I chose to email first rather than add to the issue.

mikaelpatel, 2015-02-02:
I just recently received an OLED screen and put that on the to-do-list.
Totally forgot about your work on OLED. I will have a look.

As you know I try to follow a simple principle of have hardware to test all
of the components in the Cosa repro. With an OLED at hand it is much easier
for me to test your contribution. I remember that you had a lot of
questions that went unanswered at the time. Sorry about that.

Obvious there are a lot of ripple effects when adding a new screen that is
semi-graphical. There are a lot of things to consider.

Cheers!

mikaelpatel, 2015-02-15:
@jeditekunum 
Will put a reference to your implementation in a list of libraries.

closed 2015-02-15 by mikaelpatel