mikaelpatel/Cosa #321: Recent change to miniterm.py breaks with Python 3 present

jeditekunum, 2015-02-03:
https://github.com/mikaelpatel/Cosa/pull/304

miniterm.py is pre-Python 3.  I also have Python 3 installed (from macports) so "env" picks that up since I have macports in my path ahead of system directories - intentionally.

One could go around forever arguing the wisdom of essentially two different tools with the same name - of which we have no control.  I have Python 3 installed because some packages demand it.

I know next to nothing about Python so I have no idea if there is a way this can be modified to do the right thing in any situation.  I suggest the change be rolled back and the PR originator resubmit a better solution.

mikaelpatel, 2015-02-04:
This seems to be a hen-egg problem - I cannot figure out how to make you both happy :). The "env" should be the better solution.

jeditekunum, 2015-02-04:
chicken-egg :)  Previously working status quo is better than broken and a PR submitter should be responsible for finding a portable solution...

The scripts are version 2 so they should specifically call out version 2.  It looks like every installation includes a pythonX.Y.  All the platforms I can check (OSX/Solaris/Ubuntu) have python2.7 either natively or through a supplemental package system (macports or CSW). Less common is the availability of python2 (neither OSX native or macports).  There are some situations where 2.7 wouldn't work as only 2.6 might be available.

I think Cosa scripts should specify either python2.7 or python2.  If a user doesn't have exactly what is specified and/or won't install a package that has it then it is up to them to put a link of that name somewhere in their path.  I'd lean towards:

`#!/usr/bin/env python2.7`

closed 2015-02-08 by mikaelpatel