mikaelpatel/Cosa #322: SD driver not working for my card

jeditekunum, 2015-02-03:
Trying to use [MT-SD MicroSD Card Slot with 3.3V LDO regulator and voltage translator](https://www.mattairtech.com/index.php/featured/mt-sd.html) with `CosaFAT16` example.
```
CosaFAT16: started
64:void setup():trace:free_memory() = 15278
65:void setup():trace:sizeof(FAT16::File) = 19
CosaFAT16.ino:66:assert:sd.begin(CLOCK)
```
If I insert `DELAY(600)` after [SD.cpp#L114](https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/SPI/Driver/SD.cpp#L114) it gets past that assert (and fails later which I have not debugged yet):
```
CosaFAT16: started
64:void setup():trace:free_memory() = 15278
65:void setup():trace:sizeof(FAT16::File) = 19
CosaFAT16.ino:67:assert:FAT16::begin(&sd)
```
If CLOCK is FAST_CLOCK in CosaFAT16 then the delay must be ~600 (>500).  If CLOCK is SLOW_CLOCK then the delay will work with ~500 (>400).

What is the correct way to fix this?  Seems this is dependent on the card used.

jeditekunum, 2015-02-03:
I got FAT16 working on it - it was a 4GB card formatted as FAT32 so once I reformatted for FAT16/2GB it works.  SD issue remains.

mikaelpatel, 2015-02-04:
You could also increase the retry limit https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/SPI/Driver/SD.hh#L308
This card might have a longer access sequence as it is "larger". Most cards that I have tested av small, very small, in comparison. 
Some cards seem to require that MISO is pulled low for approx 1 ms in the initialization. Have not tested this though.

jeditekunum, 2015-02-04:
https://github.com/mikaelpatel/Cosa/pull/323

closed 2015-02-04 by jeditekunum