mikaelpatel/Cosa #324: Add support for ICMP and ping messages using the W5100 and ethernet shield

sehlstrom, 2015-02-10:
Add support for the ICMP allowing to send e.g. ping messages using the W5100 and ethernet shield.

A implementation of the protocol is available for ordinary Arduino boards by the use of the ICMPPing library (http://playground.arduino.cc/Code/ICMPPing).

I will gladly contribute to the Cosa project given I get some guidance about where to start in the library. My guess is that the following is needed to be done at least:
* ICMP needs to be added to the `enum Protocol` of the Socket.hh file. 
* The function `W5100::Driver::open(Protocol proto, uint16_t port, uint8_t flag)` needs modification:
    * Setup for the the ICMP protocol using an `if`-statement
    * Status validation of the ICMP protocol
* The header file W5100.hh needs the addition of the Socket Mode Register bitfield for ICMP
* The function `W5100::Driver::recv(void* buf, size_t len, uint8_t src[4], uint16_t& port)` needs modification:
    * Add `case MR_PROTO_ICMP `

mikaelpatel, 2015-02-10:
@sehlstrom 
I think it is good to add a motivate to this. From a design point of view it would be good to reduce the changes to the driver to a minimum. The actual interface for PING should be like NTP or DNS. 

The best place to start is with the interface class for PING. Sketch an implementation that build the request message etc. After that the implication on the Socket driver and W5100 become more obvious. 

ICMP will be implemented as a RAW message for the W5100.

sehlstrom, 2015-02-10:
@mikaelpatel 
The motivation for this is to be able to ping e.g. a phone connected to the home LAN. By dedicating an IP to the MAC of the phone, you always know what adress to ping. If the phone responds to the ping, one can assume that the owner of the phone is at home. Then the Arduino can make actions based on the presence of the phone, e.g. turn on all the lights when the phone comes home or turn of the coffee machine when the phone no longer is available.

I'll start by sketching a PING interface.

And yes, I saw that the ICMPPing library sent the ping messages using the RAW message when looking deeper into the code.

mikaelpatel, 2015-02-11:
@sehlstrom 
Have you test this method of detecting presence? There might be a significant delay before the phone detects the network and "plugs-in". Also the ping frequency may reduce battery time on the phone. 

Would be good to verify the idea of presence detect and any ripple effects before doing this work. There might be a simple way to detect that someone has entered the house.

sehlstrom, 2015-02-11:
@mikaelpatel 
I have tested it in a smal scale test where a ping was sent with a rather short delay to my phone. No significant delay was observed when turning on and off flight mode on the phone simulating leaving and entering the WiFi at home. 

The test was however conducted during a short time period, so I can not say anything about the battery time on the phone. My idea has been to send the ping with some delays in order to reduce the number of pings and thereby the load on the phone(s). 

I'll setup a test logging the preens of my phones during a few days and see what happens with the battery time. Probably not much since the phones mostly is connected to the charger when at home anyway.

Another benefit of using the phone for presence detection is that you can detect who is home and base person specific actions of that.

Any other ideas of achieving the same result is happily considered. If the same information could be gathered from the router directly instead that would remove the battery issue and you can probably also increase the rate at which presence detection is done.

sehlstrom, 2015-02-11:
Started a sketch for a ICMP (ICMP.h and ICMP.cpp) implementation similar to the DHCP implementation. Not sure how to do this the best way, but it is put under my ElFi project as a branch. Is it better to make a fork of Cosa and put it there?

https://github.com/sehlstrom/ElFi/tree/ICMP/elfi

mikaelpatel, 2015-02-11:
The checksum is an excellent example of where the original RFC can be used to avoid problems. The one you have copied has a lot of quality issues. See commit https://github.com/mikaelpatel/Cosa/commit/da4d9a5b26015a0a64a03b729f2ce25c9b345b9d.

sehlstrom, 2015-02-12:
Oh! Nice :+1:

I have made a few commits to add some more to the sketch.

But I am uncertain about where to put what. The function ping(uint8_t[4] pingIp) intended to be the end user function to use, should it be in the ICMP files or in the socket drivers? Looking at how W5100 uses the begin_p-function, it appears as a ping-function actually should be placed within the W5100 and work in a similar manner.

mikaelpatel, 2015-02-12:
@sehlstrom 
I do not think it is a good idea to put too much in the driver. The structure should be something like for DHCP, that is a class ICMP with the definitions for this protocol and then more or less send() and recv(). The ping() function is achieved by send(dest, PING) and then recv(). This can then be embedded into the ping() function in ICMP. Adding it to W5100 will reduce possible reusage and there is the issue of timeout and handling of events. In some cases, it is actually important that the application can come in between the send and recv. 

The special handling of begin_P() and DHCP is mainly motivated by the mapping from symbolic names to IP addresses. Pure convenience and frequently used. Ping is not that frequent in comparison.

sehlstrom, 2015-02-12:
@mikaelpatel 
Okey. But how do I then get a free socket with the IPRAW protocol? Do I in the ICMP class retrieve one like:
```cpp
bool
ICMP::begin(Socket* sock)
{
  if (m_sock != NULL) return (false);
  m_sock = sock;
  return (true);
}

bool
ICMP::end()
{
  if (m_sock == NULL) return (false);
  m_sock->close();
  m_sock = NULL;
  return (true);
}

int
ICMP::ping(uint8_t ip[4], uint8_t retries)
{
    if (!begin(W5100::socket(Socket::IPRAW)) return (-1);
    //...
    end();
}
```
Such a solution would not make it possible to get a socket to communicate via if you have the WiFi shield instead of the ethernet.

sehlstrom, 2015-02-12:
@mikaelpatel 
Okey I think I understand.

So to get a socket to communicate via, a ICMP object is started by providing it a socket in the setup()-function. Here, a W5100 ethernet object is used, but it could be a WiFi object as well:
```cpp
icmp.begin(ethernet.socket(Socket::IPRAW));
```
Then a ping is carried out in e.g. the loop()-function:
```cpp
icmp.ping(pingIp);
```
In the ICMP::ping()-function, we call `ICMP::send(ip, ICMP_ECHO_REQUEST)` and then listen via `ICMP::recv(ip, ICMP_ECHO_REQUEST)`.

But, in ICMP::send(), I can't figure out how to send the message. I was thinking that the Socket::send()-function, by writing `m_sock->send(...)`, would do the jobb, but it is protected so I can not use it. Also, I can't figure out what the progmem flag is about in the function.

mikaelpatel, 2015-02-12:
@sehlstrom 
How about turning this around somewhat?
```
icmp.ping(ethernet.socket(Socket::IPRAW), DEST);
```
This used the intended factory design pattern.

sehlstrom, 2015-02-12:
@mikaelpatel 
Yes, that is possible, see lates commit https://github.com/sehlstrom/ElFi/commit/c8ce4e99403ac1d06e76f4eb85c031451db241f1

It dose not resolve the issue with not being able to use m_sock->send(...).

mikaelpatel, 2015-02-12:
@sehlstrom 
Thought you would discover that that is the internal driver send. There are other send() that should be used. Guess that you have looked at the W5100 header and not the Socket header. https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/Socket.hh#L238

mikaelpatel, 2015-02-12:
@sehlstrom 
Have you checked the ICMP structure and put data in network order? https://github.com/sehlstrom/ElFi/blob/ICMP/ping/ICMP.h#L84 order in message_t is not correct? header_t is last! Also the reply structure is not the actual reply send "on wire". Also you have double returns in ping.

sehlstrom, 2015-02-13:
@mikaelpatel 
Yes, I mixed the two up and missed the Socket::sned() function needed. The Cosa library is far more complex and beautiful then I am used to with multiple inherits from Socket and SPI etc.

Anyway, the code now sends my message (with order corrected). But I do not know how to debug. I would like to monitor incoming ping requests on my computer to see if the message gets through at all. But I have not found any suitable monitor software yet (using Mac OS X ). 

It appears as there is something wrong anyhow. In ICMP::recv(), I get no response at all since `m_sock->available()` returns `0`.

The reply_t struct was intended to be passed to the user holding the reply message_t as well as information about from what ip and port the respons was sent etc. But I'll see if that thought was correct later when I actually get a reply to deal with.

Double return in ping fixed.

mikaelpatel, 2015-02-13:
@sehlstrom 
The first thing is that the message is not in network order. If you check the code that you copied from the Arduino Playground there are a few functions that you have skipped.

The second thing is that the port parameter when using a raw IP socket is the IP protocol. The value zero is not a good idea. It HOPOPT IPv6 http://www.iana.org/assignments/protocol-numbers/protocol-numbers.xhtml 

There is more but let us start with this.

sehlstrom, 2015-02-13:
@mikaelpatel 
My guess is the `serialize()` and, later in `ICMP::recv()`, the `deserialize()` functions. This is where the message is coded and decoded from a structure into and from a 64 bit package, respectively.

Regarding port, it should be 1:
```cpp
/**
 * Port number to use when ICMP is the same as protocol number
 * @section Reference
 * 1. Protocol numbers
 * http://www.iana.org/assignments/protocol-numbers/protocol-numbers.xhtml
 */
 tatic const uint16_t PORT = 1;
```

mikaelpatel, 2015-02-13:
@sehlstrom 
To help you alone I have added an example sketch; https://github.com/mikaelpatel/Cosa/commit/1dd8614c413be5c422f0781c24297587a61e1d86

This will be refactored into the Cosa/INET protocols later. It was not a big deal and you did need some help :)

mikaelpatel, 2015-02-13:
Time to close shop. The demo has now been refactored to an ICMP class. See commit https://github.com/mikaelpatel/Cosa/commit/0b5bc7d4d5e321b8d24de455524c21829189d56b.

closed 2015-02-13 by mikaelpatel

sehlstrom, 2015-02-14:
@mikaelpatel 
Many thanks! It was never my intention to force you by being annoying to do this for me. But as you said, I did need some help and I am most grateful that you provided it.

mikaelpatel, 2015-02-14:
@sehlstrom 
It took awhile before I understood how far you have come in software craftsmanship. As you may see this log is now much large than the implementation :). My guess is that it would have taken even more time to guide you. It was simple easier and faster to just do it even though you where on track.

I tried to give you some hints; 1) do not cut-and-paste code, 2) avoid bad code (as the implementation you copied), 3) read, read, read, 4) write a test program first and understand the details, then and only then do the final implementation. 

I have tested this with my cell phone and it does not automatically connect to the WiFi router. Hope you are in better luck.