mikaelpatel/Cosa #325: LS7366R implementation question

kc9jud, 2015-02-14:
Hello,

I'm just starting implementing a driver for the [LSI/CSI LS7366R Quadrature Counter](http://www.lsicsi.com/pdfs/Data_Sheets/LS7366R.pdf) chip. I was curious if the architecture of Rotary::Encoder was amenable to subclassing for supporting this chip, or should I just write it as a simply another SPI driver?

mikaelpatel, 2015-02-14:
I understand that you are using Cosa in some project. I assume that you are willing to contribute to this project as this is the second issue where you are asking for support on design. The issue list is for bug reports and enhancement requests. Implementation question are assumed to be part of contributions. 

To answer your question "pro bono" again: The Rotary::Encoder class was designed for Pin Change Interrupts but could be refactored to also use this chip. There is a significant part that uses the counter value to create more abstract functionality (see https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/Rotary.hh#L196 and https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/Rotary.hh#L303). The interrupt handling for this chip is an interesting challenge if you wish to map to the Cosa Event handler.

closed 2015-02-14 by mikaelpatel

kc9jud, 2015-02-14:
Mikael,

Of course I'm planning on contributing all of these new drivers back -- I just didn't want to issue a pull request until I had an opportunity to test thoroughly. I don't want to add testing these as an additional burden for you.

Anyway, I'll look at the Encoder source tomorrow and see if it's worth the effort to implement with the LS7366R.

kc9jud, 2015-02-14:
As a side note, is there a forum where development discussions for Cosa happen? I feel like I'm abusing the issue system slightly.

mikaelpatel, 2015-02-14:
@kc9jud 

I am sorry if I got the "tone" wrong. What you bring up indirectly is an important issue about how to handle an open-source project, especially in the startup phase. I have added a few lines to the README to make the intention clearer. https://github.com/mikaelpatel/Cosa#supporting-this-project

I have limited time for other than my own development projects but try to put aside some time to package interesting parts into Cosa and make them available to others. With that said I basically have no time for a forum or even a blog. As you may have seen the Cosa blog has not been updated in over a year. It is a pity but I had to cut that to make time for other things. The topic on the Arduino forum is basically for update announcements. 

As you may have seen there are a lot of beginners that seek help. In many cases they really need to read a few books first instead of cut-and-pasting code and then asking why it does not work. 

To maintain high quality code I need to keep setups of all major hardware to allow regression testing. This has become both time consuming and actually expensive. It started in small scale but as you see become very large. Therefore I have to be very restrictive about what is put in the Cosa "core".