mikaelpatel/Cosa #330: LCD stops displaying after 2+ days run time

jhbowling, 2015-02-20:
Item | Value
------------ | -------------
Arduino IDE Version | 1.0.6
Board | Arduino UNO R3
LCD | 20x4 SainSmart LCD2004 with PCF8574T controller
Cosa Revision (last commit) | 73a9a47 Cleanup.
Port | HD44780::SainSmart

After 2+ days of runtime in the loop below, the LCD display is blank when it should be displaying data.  I can tell the loop is running because the heater is coming on and off as expected.  In addition, if I press the button (which terminates the loop), another piece of code clears the display and is able to write to the LCD, but now the cursor blinks at the last set_cursor location.  Also, is there a better way to catch a button press event (I'm missing button presses)?  I know that's more than one question but it's regarding the same piece of code.

```
 bool done = false;
 while (!done) {  
    // Rotary Encoder/Dial will push event when a change occurs
    Event event;
    Event::queue.await(&event);               
    // Dispatch the event so that the brewcoder value is updated
    event.dispatch(); 
    done = encoder->isPressed(true);
    if (!done) {
        sensor.convert_request();
        sensor.read_scratchpad();
        int16_t temp = sensor.get_temperature();
        temp = Utils::convertCtoF(temp);
        lcd->set_cursor(0, 0);
        lcd->puts_P(PSTR("Temp:"));
        char buf[10];                  
        lcd->puts(itoa(temp, buf, 10));
        if (temp <= lowMark && heaterPin->is_high()) {
            heaterPin->low();
            lcd->set_cursor(0, 1);
            lcd->puts_P(PSTR("Relay:ON "));
            trace << "turning ON...temp=" << temp << endl;                        
        } else if (temp >= highMark && heaterPin->is_low()) {
            heaterPin->high();
            lcd->set_cursor(0, 1);
            lcd->puts_P(PSTR("Relay:OFF"));                        
            trace << "turning OFF...temp=" << temp << endl; 
        }     
    }
 }
```

mikaelpatel, 2015-02-20:
Has this sketch stopped working? Did it previously work and now stopped due to updates? If so which previous version did you use? 

There are some timing issue in the sketch.

Cheers!

mikaelpatel, 2015-02-20:
@jhbowling 

More questions: How long wires are you using to the LCD? Do you have a noise source nearby? High voltage etc? 

One reason to why the LCD could go blank is that the back-light bit is turned off. This could occur due to hardware error in the transmission. It could also occur if your sketch is writing at the wrong address. As the sketch is still running I think we can rule out the event queue.

Is this a bug report on the LCD or are you simply asking for help with you sketch?

jhbowling, 2015-02-20:
Q) Has the sketch stopped working?
A) When I put the sketch into service it runs for a long time (the time it takes to ferment beer).  During development and testing I did not see this behavior.  It only occurs when the sketch has been running the loop above for an extended period of time.

Q) How long are the wires to the LCD?
A) Quite short...I would say 6 inches.

Q) Do you have a noice source nearby?  High voltate etc?
A) The sketch is running a relay that is switching 120v, so yes.

Regarding whether this is a bug report or a request for help, I would say this issue of missing button presses in the code above could be a bug if the code otherwise looks acceptable.  The issue I'm having with the LCD I'm trying to isolate at this time.  As you can see there are only a few lines of code that write to the LCD so I was looking for other potential causes.  Right now I'm suspecting the high voltage nearby.

jhbowling, 2015-02-20:
>> One reason to why the LCD could go blank is that the back-light bit is turned off.
Actually, the back-light is on, I'm just not getting characters printed to the LCD.

>>It could also occur if your sketch is writing at the wrong address.
When you say "wrong address" are you referring to row/column when calling set_cursor?

mikaelpatel, 2015-02-20:
@jhbowling 

The LCD issue seems to be a hardware problem. You need to reduce the possible noise from the relay. 

The problem with the button is a timing issue. If you check the convert_request() you will see that there is a long delay period (up to 750 ms). https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/OWI/Driver/DS18B20.cpp#L82, https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/OWI/Driver/DS18B20.hh#L282

I am assuming the you are using the Button class. In that case it requires a timeout event before sampling. https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/Button.cpp

All the details are on the table. Do you see the problem yet? :)

mikaelpatel, 2015-02-20:
@jhbowling 

BW: What did you really mean with "the LCD display is blank"? That there are no further updates? Could you please post the whole sketch (link or whatever is practical).

jhbowling, 2015-02-22:
My fundamental understanding of the event handling is not good.  I futzed with the code until I made it work, but given that you thought you were pointing me directly to the solution it took me a long time, and it may still not be what you would do. Here is what I did:
```javascript
 bool run = true;
 while (run) {         
    // Rotary Encoder/Dial will push event when a change occurs
    Event event; 
    Event::queue.await(&event);
    // Dispatch the event so that the brewcoder value is updated
    event.dispatch();
    run = !brewcoder->isPressed(true);
    if (run) {
        Watchdog::end();
        Watchdog::begin(1024, Watchdog::push_timeout_events);                        
        sensor.convert_request();
        sensor.read_scratchpad();
        Watchdog::end();
        Watchdog::begin(16, Watchdog::push_timeout_events);                       
        int16_t temp = sensor.get_temperature();
        ...remainder of loop code
```
I'm still not sure sematically what the following is doing:

`Watchdog::begin(1024, Watchdog::push_timeout_events);`

Does it mean "Watch for events in the next 1024 milliseconds and at the end of that time push them along with a timeout"?

If you want to see the code I would be happy to have you look at it but I have some cleanup to do so as not to get scolded too badly.  Maybe I'll create a GitHub project.

In answer to your question 'What did you really mean with "the LCD display is blank"?', I mean that the backlight is on, but no characters are being displayed on the LCD.  Given what the loop was doing (printing temperature and relay state), there should always be some characters being displayed on the lcd.

mikaelpatel, 2015-02-22:
Did that really help avoid missing button events? If you press and release the button during the conversion (750 ms)? Below are some solutions:

1. Process all events before checking the button state as the event queue is filling up during the wait for the sensor. How many timeout events are pending after read_scratchpad()? 

2. Alternatively put an extra event loop dispatch while waiting for the conversion to complete (between convert_request and read_scratchpad). I might need to help you with detecting that the conversion is completed as an internal timestamp is used. 

3. Alternatively making the sensor read a periodic function controlled by the button. For instance, read the sensor and then issue a convert request every second. Check the DHT device driver for an example. 

Right now the loop() is a mix of paradigms. The read_scratchpad() is blocking.

mikaelpatel, 2015-02-22:
@jhbowling 
Please see commit https://github.com/mikaelpatel/Cosa/commit/aebbc922b66fdd0ba4c82cf15e17ebf193073d55 for an example of periodic handling.

closed 2015-02-22 by mikaelpatel