mikaelpatel/Cosa #334: Q about CC3000

jeditekunum, 2015-02-23:
I tried the new code on my CC3000 (AdaFruit board).

I had to bump timeout https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/Socket/Driver/CC3000.cpp#L374 to 10000.

`wlan_connect` is returning 0 no matter what parameters I pass to it. And then it almost immediately crashes. I put a sleep(10) before the INFO("MAC line and it reboots immediately. My output is

```
CosaCC3000info: started
45:void setup():measure:Start CC3000:6069432 us
48:void setup():measure:Read software package information:256 us
51:void setup():trace:PACKAGE_ID = 1
52:void setup():trace:PACKAGE_BUILD_NR = 24

56:void setup():measure:Connect to WLAN:6321532 us
63:void setup():trace:res = 0
65:void setup():info:calling sleep(10) ...
CosaCC3000info: started
45:void setup():measure:Start CC3000:6071480 us
48:void setup():measure:Read software package information:428 us
51:void setup():trace:PACKAGE_ID = 1
52:void setup():trace:PACKAGE_BUILD_NR = 24
```

If I put traces in wlan_connect it will frequently crash before the entire trace is output.

What would you recommend for next steps in debugging? Seems like a stray interrupt but it looks like the interrupt code does nothing other than set a variable.

mikaelpatel, 2015-02-23:
@jeditekunum 
Have you tried removing the wlan_connect()? Also turn on the trace in CC3000.cpp for more info on the callbacks. BW: What Arduino are you using? Just in case you have run out of memory.

mikaelpatel, 2015-02-23:
@jeditekunum 
Please see the commit https://github.com/mikaelpatel/Cosa/commit/e1a8c5762b02209be6e980168340daa6cb8e3cd4. It enables more tracing and forces connect by using the wlan_ioctl_set_connection_policy().

jeditekunum, 2015-02-23:
Thanks. Better - not crashing now (although could be hung :)

```
CosaCC3000info: started
45:void setup():measure:Start CC3000:6069432 us
48:void setup():measure:Read software package information:428 us
51:void setup():trace:PACKAGE_ID = 1
52:void setup():trace:PACKAGE_BUILD_NR = 24

56:void setup():measure:Set connect policy:588608 us
58:void setup():trace:res = 0
60:void setup():measure:Connect to WLAN:6322716 us
67:void setup():trace:res = 0
68:void setup():info:MAC and Network addresses:
13050:HCI_EVNT_WLAN_UNSOL_KEEPALIVE:BUFFER_COUNT=6
```

If I disable the wlan_connect I get this (endless repeat of all APs):

````
CosaCC3000info: started
45:void setup():measure:Start CC3000:6069432 us
48:void setup():measure:Read software package information:256 us
51:void setup():trace:PACKAGE_ID = 1
52:void setup():trace:PACKAGE_BUILD_NR = 24

56:void setup():measure:Set connect policy:588604 us
58:void setup():trace:res = 0
70:void setup():info:MAC and Network addresses:
11006:HCI_EVNT_WLAN_UNSOL_KEEPALIVE:BUFFER_COUNT=6
MAC=70:FF:76:00:E4:B2
IP=0.0.0.0
SUBNET=0.0.0.0
DNS=0.0.0.0

88:void loop():measure:Request scan for available access points:1180376 us
91:void loop():info:Wait for scan results (10 sec)
27916:HCI_EVNT_WLAN_UNSOL_KEEPALIVE:BUFFER_COUNT=6
ssid=belkin.bbe,rssi=42,wlan_security=3
ssid=NETGEAR43,rssi=42,wlan_security=3
ssid=kingofthejungle-guest,rssi=37,wlan_security=0
ssid=kingofthejungle,rssi=35,wlan_security=3
ssid=HP-Print-45-LaserJet 200,rssi=42,wlan_security=3
ssid=belkin.e1c,rssi=35,wlan_security=3
ssid=Hyperspace,rssi=72,wlan_security=3
ssid=Maas,rssi=39,wlan_security=3
ssid=Sonbol,rssi=43,wlan_security=3
ssid=anderson0a,rssi=34,wlan_security=3
ssid=belkin.bbe.guests,rssi=41,wlan_security=0
ssid=NETGEAR56e.guests,rssi=39,wlan_security=3
ssid=Hyperspace-Guest,rssi=83,wlan_security=3
ssid=NETGEAR65,rssi=41,wlan_security=3
ssid=Schaeffer,rssi=33,wlan_security=3
ssid=Schaeffer-guest,rssi=31,wlan_security=0
ssid=9b826a,rssi=33,wlan_security=3
ssid=NETGEAR,rssi=35,wlan_security=0
ssid=HoleInOnea-guest,rssi=39,wlan_security=3
ssid=NETGEAR99,rssi=33,wlan_security=3
ssid=belkin.bbe,rssi=42,wlan_security=3
ssid=NETGEAR43,rssi=42,wlan_security=3
ssid=kingofthejungle-guest,rssi=37,wlan_security=0
ssid=kingofthejungle,rssi=35,wlan_security=3
...
```

I'll look further tomorrow. Device is MoteinoMEGA so lots of memory.

jeditekunum, 2015-02-24:
Odd. Today its crashing again, with or without wlan_connect.

```
CosaCC3000info: started
45:void setup():measure:Start CC3000:6069432 us
48:void setup():measure:Read software package information:428 us
51:void setup():trace:PACKAGE_ID = 1
52:void setup():trace:PACKAGE_BUILD_NR = 24

56:void setup():measure:Set connect policy:584764 us
58:void setup():trace:res = 0
60:void setup():measure:Connect to WLAN:6321572 us
65:void setup():trace:res = 0
67:void setup():info:MAC and Network addresses:
13045:HCI_EVNT_WLAN_UNSOL_KEEPALIVE:BUFFER_COUNT=6
CosaCC3000info: started
45:void setup():measure:Start CC3000:6071480 us
48:void setup():measure:Read software package information:256 us
51:void setup():trace:PACKAGE_ID = 1
52:void setup():trace:PACKAGE_BUILD_NR = 24

56:void setup():measure:Set connect policy:102644 us
58:void setup():trace:res = 0
60:void setup():measure:Connect to WLAN:6321568 us
65:void setup():trace:res = 0
67:void setup():info:MAC and Network addresses:
12565:HCI_EVNT_WLAN_UNSOL_KEEPALIVE:BUFFER_COUNT=6
CosaCC3000info: started
```

Its immediate after the KEEPALIVE event.

jeditekunum, 2015-02-24:
If I put a `sleep(1);` in https://github.com/mikaelpatel/Cosa/blob/master/examples/WiFi/CosaCC3000info/CosaCC3000info.ino#L43 then begin_P never returns. If I put a `delay(1);` in there it does. No idea if this has anything to do with it.

jeditekunum, 2015-02-24:
I added a bunch of traces, bumped begin_P timeout. It looks like it is talking to my device but the device is not sending back HCI_EVNT_WLAN_UNSOL_CONNECT, just KEEPALIVE.

```
CosaCC3000info: started
258:bool CC3000::begin_P(str_P, uint16_t):trace:timeout = 10000
553:int CC3000::simple_link_start(uint8_t):trace:res = 1
554:int CC3000::simple_link_start(uint8_t):trace:evnt.status = 0
567:int CC3000::read_buffer_size(uint8_t&, uint16_t&):trace:res = 4
568:int CC3000::read_buffer_size(uint8_t&, uint16_t&):trace:evnt.status = 0
278:bool CC3000::begin_P(str_P, uint16_t):trace:BUFFER_COUNT_MAX = 6
279:bool CC3000::begin_P(str_P, uint16_t):trace:BUFFER_MAX = 1500
11006:HCI_EVNT_WLAN_UNSOL_KEEPALIVE:BUFFER_COUNT=6
298:bool CC3000::begin_P(str_P, uint16_t):trace:res = -62
299:bool CC3000::begin_P(str_P, uint16_t):trace:evnt.status = -2
510:int CC3000::nvmem_read(uint8_t, void*, uint32_t, size_t):trace:res = 1
45:void setup():measure:Start CC3000:11159180 us
583:int CC3000::read_sp_version(uint8_t&, uint8_t&):trace:res = 5
584:int CC3000::read_sp_version(uint8_t&, uint8_t&):trace:evnt.status = 0
48:void setup():measure:Read software package information:14920 us
51:void setup():trace:PACKAGE_ID = 1
52:void setup():trace:PACKAGE_BUILD_NR = 24

477:int CC3000::wlan_ioctl_set_connection_policy(bool, bool, bool):trace:res = 5
478:int CC3000::wlan_ioctl_set_connection_policy(bool, bool, bool):trace:evnt.status = 0
479:int CC3000::wlan_ioctl_set_connection_policy(bool, bool, bool):trace:evnt.result = 0
56:void setup():measure:Set connect policy:139564 us
58:void setup():trace:res = 0
389:int CC3000::wlan_connect(CC3000::Security, str_P, str_P, str_P):trace:res = 5
390:int CC3000::wlan_connect(CC3000::Security, str_P, str_P, str_P):trace:evnt.status = 0
391:int CC3000::wlan_connect(CC3000::Security, str_P, str_P, str_P):trace:evnt.result = 0
60:void setup():measure:Connect to WLAN:6358904 us
65:void setup():trace:res = 0
66:void setup():info:MAC and Network addresses:
CosaCC3000info: started
```

When working with "devices" I like to have the ability to trace every command sent and every response received, human formatted (for example, symbolic names for commands and events). ```issue``` and ```await``` would be good places but they are in HCI and the event definitions are in CC3000.hh. Not sure how you would prefer; could CC3000:: have wrappers for issue/await?

mikaelpatel, 2015-02-25:
@jeditekunum 
Changed to another WiFi router and go a lot of timing issues. Especially the DHCP event was very much delayed. Have tested another strategy for the startup. Please see the commit https://github.com/mikaelpatel/Cosa/commit/cca0cd42d9143bbc4e2def3cbab2f4534d41eb45.

jeditekunum, 2015-02-25:
With `wlan_connect` it almost never produces any scan results. While trying a bunch of things over the last couple of hours I saw a DHCP event exactly ONCE.

```
CC3000info: started
44:void setup():measure:Start CC3000:1271992 us
47:void setup():measure:Read software package information:260 us
50:void setup():trace:PACKAGE_ID = 1
51:void setup():trace:PACKAGE_BUILD_NR = 24

67:void setup():measure:Connect to WLAN:6351996 us
75:void setup():info:MAC and Network addresses:
MAC=70:FF:76:00:E4:B2
IP=0.0.0.0
SUBNET=0.0.0.0
DNS=0.0.0.0

94:void loop():measure:Request scan for available access points:691668 us
97:void loop():info:Wait for scan results (10 sec)
18399:HCI_EVNT_WLAN_UNSOL_CONNECT:m_buffer_avail=6
18400:HCI_EVNT_WLAN_UNSOL_DHCP:m_buffer_avail=6
18406:HCI_EVNT_WLAN_UNSOL_KEEPALIVE:m_buffer_avail=6
ssid=chriskarla,rssi=39,wlan_security=3
ssid=chriskarla-guest,rssi=39,wlan_security=0
ssid=MJCACB,rssi=35,wlan_security=2
ssid=NETGEAR43,rssi=43,wlan_security=3
ssid=kingofthejungle-guest,rssi=34,wlan_security=0
...
```

If I remove `wlan_connect` then I consistently get scan results. But most the time  `wlan_ioctl_get_scan_results` repeats forever. Rarely, usually on the first time in `loop` it will show a good set of scan results and then the next time into `loop` it repeats results for awhile and then dies with

```
...
ssid=Galaxy,rssi=36,wlan_security=1
ssid=chriskarla,rssi=40,wlan_security=3
101:void loop():measure:Get scan results:6608892 us

CosaCC3000info.ino:95:assert:wifi.wlan_ioctl_set_scanparam() == 0
```

Does getting scan results from the device automatically stop it from scanning? If not, perhaps it needs to be told to stop scanning after the 10 second wait and before the results are retrieved.

My available APs are Netgear R7000 and Apple Airport Extreme; both WPA2. Behavior seems to be similar.

jeditekunum, 2015-03-03:
I'm giving up on CC3000.

closed 2015-03-03 by jeditekunum

mikaelpatel, 2015-03-03:
I have also pushed it off the backlog. There are more fun things to do than try to get it working. The  event stream is just too asynchronous. And it sometime takes for ever to connect to the network.