mikaelpatel/Cosa #337: Compatibility with "ino" build tool

jgarbers, 2015-03-11:
I'm not sure if this would be a feature to add to *ino* or one for Cosa, but I'd like to be able to build and deploy Cosa-based apps from the command line using the "ino" command line build tool.  I can use regular Arduino libraries by copying them into my ion-based project's "lib" directory, but I'm not sure how I would incorporate Cosa.  Does anyone with more experience with these tools have any ideas as to how to get started? Thanks!

mikaelpatel, 2015-03-11:
@jgarbers 
Cosa is a totally different approach to programming the Arduino (and AVR). Combining the two cores is tough or even impossible for the average Arduino user. 

What libraries or functionality do you need? That might already be available in Cosa. 

Cheers!

jgarbers, 2015-03-11:
Hi, and thanks for your answer! I'm not trying to blend the cores at all... I would just like to use a command-line-based build process when working with Cosa.  The "ino" tool I referred to doesn't appear to support "alternative" cores, so I suspect the limitation is on its side.

I can build and run Cosa-based code from UECIDE and the 1.6.1 Arduino IDE... do you know of a way to do builds and uploads using Cosa from the command line? I'm on a Mac, if that matters.

Once I get that up and running, I'd like to use some of the Adafruit libraries -- in particular, their support for the NeoPixel LED strips.  Is there an equivalent under Cosa?

mikaelpatel, 2015-03-11:
@jgarbers 
There is a command-line based build already in Cosa. It is a set of default make rules together with some shell scripts. This allows build with make but without makefiles. I have not tested it on a Mac. But that has buildutil and a shell so that should not be a problem. Please see the directory build. There are a few path setups but it is very straight forward. Some install info in https://github.com/mikaelpatel/Cosa/blob/master/INSTALL.txt

No, I have not added support for LED strips. "Still in the box". As Cosa pin functions are X2-X10 faster than Arduino I was hoping to implement that directly without any extra assemble etc.

jeditekunum, 2015-03-11:
@jgarbers
I use the command line all the time on Mac.

I actually use my own script which I uncreatively called *avr*.  It tries to figure out upload device automatically.  See https://dl.dropboxusercontent.com/u/103345528/avr  You would have to adjust for your environment, OBJS variable for example.

jgarbers, 2015-03-11:
@mikaelpatel @jeditekunum - thanks so much for your help! I have both the *cosa* and *avr* commands working -- mostly! *avr upload* doesn't seem to pick up the proper serial port for my Uno; although it reports 

    - [AUTODETECTED]       DEVICE_PATH = /dev/tty.usbmodem411 

it winds up with the error

    No /dev/tty.usbserial* found!

Not sure what's going on there, but I can bludgeon the *avr* script into finding the right port here.  

Next step is to modify the Vim plugin I use to invoke *ino* and have it invoke *cosa* or *avr* instead.

Thanks again!

jeditekunum, 2015-03-11:
@jgarbers 
*avr* looks in boards.txt for the bootloader.path= for your board.  If it sees 'caterina' bootloader then it assumes it will be a "modem" USB device; otherwise it defaults to "serial" USB device.  Just add whatever your bootloader looks like to that case statement and pick "modem".

Its a quick kludge so I only put in stuff that was needed for my limited board collection.

These scripts are just front-ends to the Makefile that is somewhat limited.  It can automatically find stuff - to a point - but it doesn't work if you have other devices that match the */dev/tty.usbserial** or */dev/tty.usbmodem** patterns.  For example, I have a BitScope that appears like that.

jgarbers, 2015-03-11:
@jeditekunum Thanks for the clarification. For the moment I just set 'em all to use "modem", but I'll note your advice in case I pull out another board that looks like a "serial".  

In the *boards.txt* that comes with Cosa, the Uno section doesn't have a *bootloader.path" -- should it be "caterina"? I'm a bit out of my depth here.

jeditekunum, 2015-03-11:
@jgarbers 
Actually, bootloader is not a good way to do this.  The Uno has a builtin USB to serial converter so the processor itself is only serial and thus doesn't have a USB bootloader.  And that builtin converter exposes itself as a modem rather than serial.  External USB to serial converters, like the FTDI products, tend to expose themselves as serial.

Maybe @mikaelpatel can suggest another way to figure out if a board is connected via modem or serial device.

mikaelpatel, 2015-03-12:
@jeditekunum @jeditekunum 
Would the vid/pid attribute be useful for detecting modem/serial on Mac? They are only available on AVR devices with USB support (ATmega32U4). Unfortunately I have not included the bootloader.path on all boads. This is just to avoid having bootloaders in Cosa.

jeditekunum, 2015-03-12:
@mikaelpatel 
Every board that has a vid/pid also declares boatloader.path to be caterina.  The issue is boards that have integrated USB to serial converters that present as a modem, like Uno.  I suppose there are also external USB to serial converters that also present as a modem.  There probably isn't a good way to automate this.

On Mac the device name ends with the USB device serial number.  For example, I have a FTDI-based adapter that has a device name */dev/tty.usbserial-A703X8PQ* and the driver info shows that *A703X8PQ* is the serial number reported by the device.

I guess the easiest thing to do would be for a script to just have a list of known USB device serial numbers that are used for AVR work.  Users would have to update the list for their devices.  I suppose the list could be external to the script in a *~/.cosadevices* file (or something else).

At any rate, *avr* was just a quick hack to get around deficiencies in the *Cosa/build/Arduino-Makefile/Arduino.mk* file (which originated outside Cosa).  The auto-detect in there just uses a wildcard for any *tty.usbserial** or *tty.usbmodem** which only works in the simplest of cases.  In my case using a BitScope to debug something broke that since it is also a tty.usbserial.

jgarbers, 2015-03-12:
This is probably naive, but aren't USB devices generally supposed to be self-identifying? If I run

    system_profiler SPUSBDataType

I see an entry like

       Arduino Uno:

          Product ID: 0x0001
          Vendor ID: 0x2341
          Version: 0.01
          Serial Number: 64934333235351C09051
          Speed: Up to 12 Mb/sec
          Manufacturer: Arduino (www.arduino.cc)
          Location ID: 0x04100000 / 4
          Current Available (mA): 500
          Current Required (mA): 100

Would identifying an Arduino device based on info like this be possible? I don't know if the USB info is available via any pattern-match-friendly mechanism or not. The output from *system_profiler* looks mostly human-friendly... there's an XML output option, but parsing that would probably be more trouble than it's worth.

jeditekunum, 2015-03-12:
@jgarbers 
That is a long serial number!  Does your device name end with all of that?

The Uno actually has 2 AVR processors on it.  The Atmega328 that you program and an Atmega16U2 that has been programmed by the manufacturer to be a USB to serial converter.  The converter is not designed to be programmed by the user.

Other devices have a native USB interface.  For example, the Leonardo is an Atmega32U4.  In that case the bootloader and then application code implement the USB protocols directly.  In Cosa you can look at `cores/cosa/Cosa/USB/*` to see the USB implementation when Cosa applications are running.

All that stuff you are seeing is just the USB device software presenting a product name ("Arduino Uno" in your case), product id, vendor id, serial number, etc.  The vendor id is assigned by the USB Implementors Forum (www.usb.org) after payment of substantial $$$.

It is the vendor and product IDs that map to drivers on the host.  The driver implementation determines whether its a modem or serial device.

With true Arduino manufactured devices you can probably depend on the name ("Arduino Uno" for example) to find the serial number which then presumably appears in the device name (so you could match `/dev/tty.usb*-$SERIAL`).

Many of us don't have Arduino manufactured devices.  Lots of boards are also just serial so they don't present anything useful.

For example, this is what one of my FDTI adapter shows:

```
    FT232R USB UART:

      Product ID: 0x6001
      Vendor ID: 0x0403  (Future Technology Devices International Limited)
      Version:  6.00
      Serial Number: A703X8PQ
      Speed: Up to 12 Mb/sec
      Manufacturer: FTDI
      Location ID: 0xfd330000 / 9
      Current Available (mA): 500
      Current Required (mA): 90
```

It could be connected to anything.

Sometimes I have one cable for programming and another for talking serial on another port to the same device - or another device.

Sorry for the long comment.  In the end one can only automate the really simple situations.

jgarbers, 2015-03-13:
@jeditekunum Thank you for such a detailed explanation!  That helps a lot.

My device shows up as */dev/tty.usbmodem411".

One approach to the problem might be to allow an environment variable, say *ARDUINO_PORTS*, that would be a colon-delimited list of ports that have Arduino devices connected to them -- basically like *PATH*.  The script would choose the first one in that list that existed on the system.  That way I can just add new values to *ARDUINO_PORTS* as I get new devices, and if I want to use more than one at a time, rearrange the values to reflect my priority.

jeditekunum, 2015-03-13:
@jgarbers 
Hmmm... that device name looks familiar.  I just remembered that I have removed the Apple driver and replaced with the FTDI driver because the Apple one has problems (this is on Mavericks).  That is why my names have the serial number in them.  Sorry, I had completely forgot about that.  If I recall correctly the Apple driver naming is based on what USB socket the device is plugged into; I think if you try a different socket your name will change!

Not saying you need to replace the driver but if you want to there are lots of places that describe how to do it.  [Here](http://forum.arduino.cc/index.php?topic=198539.0) is one.

Yep, the device list is a possible solution.  I'll take a look at makefile to see how hard that would be; no promises ;)  @mikaelpatel please chime in with any opinion on this...

jgarbers, 2015-03-13:
At the risk of drifting from the original topic... thanks for the tip on the FTDI drivers. I'm running Yosemite, and the only issue I've encountered is that some boards won't work when connected via a hub, but they're fine with plugged in to one of the Mac's own ports.  (Wonder what'll happen with the new one-port MacBook?)  I'll bookmark the tips you linked to, but it looks like the FTDI drivers for Mac haven't been updated since 2012, so I hesitate to try those on Yosemite until I have a worse problem.  Thanks!

jeditekunum, 2015-03-13:
@jgarbers @mikaelpatel 
Turns out this is easy.  Please see changes I propose https://github.com/jeditekunum/Cosa/commit/61d9b47b22113bed0b2c6abbac29f509c90e156a

If the environment variable MONITOR_PORTS is set then it is assumed to be a ':' separated list of device paths that may be valid avr devices.  The wildcard auto detect will then be limited to those in that list.  As with no MONITOR_PORTS set, if there are more than one present the first is taken.

I propose MONITOR_PORTS rather than ARDUINO_PORTS because its consistent with Cosa and the alternative implies something supported by Arduino which it isn't.

To go with this I've altered my *avr* script, temporarily called *avr2*, which can be found [here](https://dl.dropboxusercontent.com/u/103345528/avr2).

Please review/comment and if acceptable to both of you I will submit a PR to Mikael.

jgarbers, 2015-03-13:
I can't offer a qualified opinion on the code itself, but the approach sounds great. Thumbs up from here.

mikaelpatel, 2015-03-13:
I think that was a great solution. Nice and simple :).

jeditekunum, 2015-03-13:
@jgarbers Re: *avr2 port*
It should display the devices found that match your MONITOR_PORTS setting. All it does is step through each path in MONITOR_PORTS and check to see if it exists and is a character special file (the *-c* test).  If found it just echoes the path.

To debug you could add a line before `for PORT in` in the port argument handling that is
`set -x`

I submitted PR https://github.com/mikaelpatel/Cosa/pull/339 for the makefile changes.

jeditekunum, 2015-03-14:
@jgarbers 
I should work to eliminate the need for the *avr/avr2* scripts.

I originally added *port* goal because I have ARDUINO_QUIET set and when that is set there is no way for the makefile to show the port/device. Looking into this again I see a simple solution that I will submit as a PR. When the goal is 'config' the makefile should ignore the ARDUINO_QUIET setting. This way the 'config' goal will always show the config including the DEVICE_PATH :)

I will continue to work to enhance the makefile so that none of the features of my script are needed.

jeditekunum, 2015-03-15:
Ok, after latest PR the *avr* scripts should be obsolete.  The *port* functionality is now provided, regardless of ARDUINO_QUIET setting, by the *config* goal.  I also enhanced *avanti* so that it will now wait awhile for true USB-based AVRs to reboot before trying to monitor.  I've removed the scripts from my Dropbox.  Please use the *cosa* script in the build directory.

As a side, the one thing that I would like to see done differently regarding the *cosa* script is the need to supply the board type all the time.  Really wish this could default to BOARD_TAG environment variable.

jgarbers, 2015-03-15:
The need to specify the board each time struck me as odd too, but I'm too pleased with Cosa to bring it up. But as long as @jeditekunum mentioned it... :-)

mikaelpatel, 2015-03-16:
You could always try a shell alias :)

BW is it time to close this issue. I think we have detoured a number of times.

jgarbers, 2015-03-16:
An alias certainly solves the problem. Thanks for all the help and information here!

closed 2015-03-16 by jgarbers

mikaelpatel, 2015-03-16:
Special thanks to @jeditekunum for the updates. Cheers!