mikaelpatel/Cosa #341: Burn fuse/bootloader for ATtinyXX on OSX

mikaelpatel, 2015-03-16:
Would appreciate if someone could test burning fuse/bootloader for ATtinyXX on OSX. There is an updated avdude.conf file that works fine for Linux. https://github.com/mikaelpatel/Cosa/blob/master/build/avrdude.conf 

There is two ways to test; 1) use the Arduino IDE and select the Cosa ATtinyXX board and then select Burn Bootloader, or 2) use the Cosa command line build (see command list).

dansut, 2015-03-16:
I have an Adafruit Trinket 5V (ATtiny85) that has never even come out of its packet, and OSX - so maybe I can help, although I have to admit I have not messed with bootloaders at all yet in my playing with AVR stuff so I might need a bit of handholding...

Did I ought to somehow backup the bootloader that is currently on it? Adafruit seem to suggest it is slightly special on their page, although I'm sure it is downloadable if I need to return to this at some time anyway.

jgarbers, 2015-03-16:
I have some Trinkets too -- set aside for a wearables project for my wife, who won't be happy if I brick them inadvertently!  Will watch here for instructions and guidance.

mikaelpatel, 2015-03-16:
Thanks for the support but I would not try it on those. They will be bricked. The Cosa bootloader for ATtinyXX is not really a bootloader. It is fuse setting (8 MHz internal clock) and then the blink sketch thats all. Still need an ISP.

dansut, 2015-03-16:
They do give instructions for repairing the Trinket bootloader using an Uno as the ISP: https://learn.adafruit.com/introducing-trinket/repairing-bootloader
If I can find time I wouldn't mind playing about with this.

If I understand this correctly though we would be able to burn the Cosa bootloader to the Trinket exactly once without an external ISP - its original bootloader acts like a USBtinyISP until we write over it. I have on my list to get an ISP but have not researched yet really what will work well for me - ultimate goal is to burn a bootloader into ATmega1284p... Any suggestions would be welcome.

mikaelpatel, 2015-03-16:
I would guess that Cosa sketches could be uploaded directly on the Trinket. It is a question of changing the boards.txt file with the correct attributes for the bootloader type. 

What I was looking for was testing ATtiny85 fuse burn and sketch upload with an external ISP on OSX. 

The ATmega1284p is already support with two different bootloaders. Though I need to add the bootloader code (hex) or at least give a reference to where they can be found. Again an external ISP is needed. The easiest way is to use an Arduino as an ISP. There is one in the Arduino library but also a Cosa version in tools.

closed 2015-05-15 by mikaelpatel