mikaelpatel/Cosa #343: Keep USB alive after an ASSERT failure

jeditekunum, 2015-03-23:
Failed ASSERT ends up calling `exit` via `trace.fatal`.  As I understand it, Arduino exit means disable interrupts and go into an infinite loop.  This prevents downloads without a reset button press.

When an ASSERT fails it should loop handling USB events so that the baud-shift-driven transfer to the bootloader still works.

Not sure of best way to implement this otherwise I would have PR.

mikaelpatel, 2015-03-24:
One solution is to remove the exit() from fatal and just use a loop. This would work for when trace is included but the problem is the same when CDC is not used at all in ATmega32U4 based variants. 

I would also guess that there are still some timing issue when trying to reconnect to a CDC based device. It is difficult to do that gracefully.

jeditekunum, 2015-03-24:
What do you think of this solution?  Obviously can't do anything about case where CDC isn't used.

https://github.com/jeditekunum/Cosa/commit/3b461d5ffb44f23743f52b2b7ae35d33906f247e

This seems to be completely reliable in my testing.  Interrupts are still disabled so no user code would run.  Takes care of any exit, not just assert.

With `ard-reset-arduino`, and recently `ard-wait-arduino`, watching for inode change I think most of the timing issues are gone.

mikaelpatel, 2015-03-24:
I have test simple removing exit() from fatal and replacing it with an infinite loop. It seems to work fine as well. The behavior would be the same for all boards, e.g., the interrupt handlers are not disabled. There is only a print of the EXIT character followed by flush and the infinite loop.

jeditekunum, 2015-03-24:
Opinions:
* bootloading should work whenever possible - including if the application has called `exit`
* `exit` should terminate the application; since the application could include interrupt handling I think that should be disabled

Implementing `exit` brings up another question... wouldn't it be desirable to print a string and then the EXIT character whenever `exit` was called?  Something like `**exited**`.

jeditekunum, 2015-03-26:
Please see PR https://github.com/mikaelpatel/Cosa/pull/347

The message on exit needs to be deferred due to issue https://github.com/mikaelpatel/Cosa/issues/346

closed 2015-03-26 by jeditekunum