mikaelpatel/Cosa #346: trace global not always initialized

jeditekunum, 2015-03-26:
This is likely a compiler issue (possibly long since corrected in newer GCC; I'm using Arduino 1.5.8)...

When implementing https://github.com/mikaelpatel/Cosa/pull/345 I discovered that the trace m_dev was not always NULL initially using this test code:

```
#include "Cosa/IOStream/Driver/UART.hh"
#include "Cosa/Trace.hh"

IOStream ios(&uart);

void setup()
{
  uart.begin();
  ios << PSTR("--> started") << endl << flush;

  if (trace.get_device() != NULL)
    {
      ios << PSTR("WARNING: trace.get_device()=") << trace.get_device()
          << PSTR(", forcing to NULL") << endl << flush;
      trace.end();
    }

#if 0
  trace.begin(&uart);
#endif

  trace.print(7); trace.println(); trace.flush();
  ios << PSTR("--> after trace.print(7)") << endl << flush;

  trace << PSTR("hello from trace") << endl << flush;
  ios << PSTR("--> after trace <<") << endl << flush;
}

void loop()
{
}
```

It appears that the constructor is not always being handled correctly.  This appears to be related to trace global being declared as weak.

mikaelpatel, 2015-03-30:
Does it work with this small change? https://github.com/mikaelpatel/Cosa/commit/3c1d7d77dacaf1b8cccdf99ffe7d50fefe13f0c9

I simply moved the attribute to the definition instead.

jeditekunum, 2015-03-30:
I am no longer able to reproduce the problem. Thanks!

closed 2015-03-30 by jeditekunum