mikaelpatel/Cosa #349: Add support for boards based on ARM/SAM

mikaelpatel, 2015-03-29:
This is a major port of Cosa to another processor architecture. There are possible ripple effects of adding a new level of variant handling if the same source tree should be used. 

There is a small set of Cosa classes that actually access AVR registers and modules. These will be reimplemented for the ARM/SAM. There are issues with timing, low-power and interrupt levels. 

The Cosa design for low-power delay, etc, is based on the Watchdog to allow power-down sleep.

mikaelpatel, 2015-04-01:
The current design is to refactor classes that use the AVR registers and specific code into the AVR directory with a mapping file such as Boards.hh. A new direction for SAM would be introduced with the SAM implementations. 

Typically classes that will be refactored are; Pin, InputPin, OutputPin, etc for Pins, UART, SPI, TWI and Timer related classes. All other classes will become common. 

A further refactoring is to better support base data types and allow the compiler to do most of the decision making. The AVR code is heavily optimized for int8_t/uint8_t where in some cases it is better to allow int/unsigned int. Mapping between uint16_t and void* is not valid on SAM without adjustment (displacement).

ODwyerPW, 2015-08-20:
I like the idea of Cosa support for Arduino ZERO (Cortex-M0+), Arduino DUE (Cortex-M3) and perhaps even TI TM4C1294 (Cortex-M4F). I happen to be playing with the TM4C1294 board quite a bit these days.

closed 2015-09-20 by mikaelpatel