mikaelpatel/Cosa #354: Problem with multiple Timers

jeditekunum, 2015-04-10:
Not sure how to categorize this. I've reproduced in the "small" test case that follows. I have two "periodic" timers that are derived from Timer. One is firing "fast" (1 ms) and the other "slow" (100 ms). If there is too big a delay in the event processing loop the slower timer never goes into `on_event`; `run` is executed normally.

```
// For me, "beat2" never does on_event when LOOP_DELAY_MS > 7

#define LOOP_DELAY_MS 8
#define BEAT1_INTERVAL_MS 1
#define BEAT2_INTERVAL_MS 100

#include "Cosa/IOStream/Driver/UART.hh"
#include "Cosa/Trace.hh"

// ======================================================================
// PeriodicTimer.hh

#include "Cosa/Timer.hh"

/**
 * Periodic function handler based on Timer (for accuracy).  Allows
 * periodic activities to be scheduled with a milliseconds level
 * resolution.  Requires RTC and Timer.  Subclass and implement
 * the virtual methods run() and/or on_event().  run() executes
 * within the interrupt context and on_event() can be used for
 * non-interrupt context.
 */

class PeriodicTimer : private Timer {
public:
  /**
   * Construct a periodic function handler based on Timer.
   * @param[in] ms period of timeout.
   */
  PeriodicTimer(uint16_t ms) : m_period(ms), m_last_start(0) {}

  /**
   * Set timeout period.
   * @param[in] ms period of timeout.
   */
  void set_period(uint16_t ms)
  {
    m_period = ms;
  }

  /**
   * Start the periodic function.
   */
  virtual void begin();

  /**
   * Stop the periodic function.
   */
  virtual void end();

  /**
   * @override PeriodicTimer
   * The default null function.
   */
  virtual void run() {}

  /**
   * @override Event::Handler
   * Periodic event handler.
   * @param[in] type the type of event.
   * @param[in] value the event value.
   */
  virtual void on_event(uint8_t type, uint16_t value)
  {
    UNUSED(type);
    UNUSED(value);
  }

private:
  uint16_t m_period;
  uint32_t m_last_start;

  /**
   * @override Timer
   * Timer member function that is called when the timeout period
   * has expired. This member function must be overridden.
   */
  virtual void on_expired();

};

// ======================================================================
// PeriodicTimer.cpp

#define US_PER_MILLIS 1000UL

void
PeriodicTimer::begin(void)
{
  Timer::begin();
  attach(this);

  m_last_start = RTC::micros();
  expire_at(m_last_start + m_period*US_PER_MILLIS);
  start();
}

void
PeriodicTimer::end(void)
{
  stop();
}

void
PeriodicTimer::on_expired(void)
{
  m_last_start += m_period*US_PER_MILLIS;
  expire_at(m_last_start + m_period*US_PER_MILLIS);
  start();
  run();
  Event::push(Event::TIMEOUT_TYPE, this);
}

// ======================================================================

class Beat1 : public PeriodicTimer {
public:
  Beat1() :
    PeriodicTimer(BEAT1_INTERVAL_MS)
  {
  }
  virtual void on_event(uint8_t type, uint16_t value);
};

void
Beat1::on_event(uint8_t type, uint16_t value)
{
  UNUSED(value);
  ASSERT(type == Event::TIMEOUT_TYPE);

  //  trace << '.';
}

static Beat1 beat1;

class Beat2 : public PeriodicTimer {
public:
  Beat2() :
    PeriodicTimer(BEAT2_INTERVAL_MS)
  {
  }
  virtual void run();
  virtual void on_event(uint8_t type, uint16_t value);
};

void
Beat2::run(void)
{
  trace << '!';
}

void
Beat2::on_event(uint8_t type, uint16_t value)
{
  UNUSED(value);
  ASSERT(type == Event::TIMEOUT_TYPE);

  trace << '*';
}

static Beat2 beat2;

// ======================================================================

void setup()
{
  RTC::begin();
  Timer::begin();

  uart.begin();
  trace.begin(&uart);

  TRACE(LOOP_DELAY_MS);
  TRACE(BEAT1_INTERVAL_MS);
  TRACE(BEAT2_INTERVAL_MS);

  beat1.begin();
  beat2.begin();
}

void loop()
{
  Event event;
  Event::queue.await(&event);
  event.dispatch();

  delay(LOOP_DELAY_MS);
}
```

jeditekunum, 2015-04-12:
User error.

Instead of
``` 
  Event::queue.await(&event);
  event.dispatch();
```
which was only processing 1 event per LOOP_DELAY_MS, this is correct:
```
  while (Event::queue.dequeue(&event))
    event.dispatch();
````
which will process all queued events.

closed 2015-04-12 by jeditekunum

mikaelpatel, 2015-04-12:
Might need to update Event::service() as it only dispatches a single event. Might be a source of error.