mikaelpatel/Cosa #365: Package Cosa for Arduino Boards Manager

tht, 2015-06-03:
With Arduino 1.6.4 there is a "Boards Manager" which allows to include third-party boards into the Arduino IDE.

There are already many third-party boards supported (https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls). It would be great if Cosa is also packaged as such an extension to allow easier installation.

Some documentation:
- https://github.com/arduino/Arduino/wiki/Arduino-Hardware-Cores-migration-guide-from-1.0-to-1.6
- https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification

mikaelpatel, 2015-06-03:
That would be possible though there are some issues. 

The first issues is that Cosa is not really an Arduino core as it does not implement the Arduino API. It is an alternative OOP Framework that runs on Arduino and clone boards. 

The second issue is that there are no Cosa specific boards. AVR Arduino boards are supported and a bunch of clones. 

I would be pleased to add the necessary JSON property file(s) but it would be nice to know that it actually would be added to the Boards Manager list. Otherwise this work is fruitless.

tht, 2015-06-03:
I've used Cosa before so I do understand that Cosa is not a "Board". For me it looks like it's implemented like a "Board" as it's also available in the Boards-Selection of the Arduino IDE but I do understand your considerations. I don't know if it's even possible to convert Cosa into such an "Add-On".

As for your second issue… Actually everyone has to add the URLs to the third-party boards he/she likes to a text-field in the Arduino IDE preferences window before the "Boards Manager" shows them. There are only very few Arduino-branded boards in this "Boards Manager" by default. I've already added tree URLs to this text-field to get all the boards I use in there. I can't tell if they would allow Cosa to be added to the "Unofficial list of 3rd party boards support urls" I've posted before. But I can tell for sure that people will be able to add it manually to their Arduino IDE.

So what's your benefit in releasing such a JSON-file? An installation which is a little bit easier and it's also easier to keep Cosa up-to-date (follow the official releases, not follow all the git commits). Maybe this helps to get a little bit more attention to your framework.

Thanks for considering it.

mikaelpatel, 2015-06-04:
Please give the Cosa package board manager file a test; https://raw.githubusercontent.com/mikaelpatel/Cosa/master/package_cosa_index.json

Cheers!

mikaelpatel, 2015-06-04:
Updated the list as well https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls

dgschwend, 2015-06-04:
I get a NullPointerException when trying to update the board manager (after adding the JSON file URL). Cannot restart Arduino after that, first need to manually delete `package_cosa_index.json` in ~/Library/Arduino15. I'm on Mac OSX 10.10.3 with Arduino 1.6.4.

tht, 2015-06-04:
That's awesome. Thank you very much. I've just installed it and did compile (and upload) some of your example programs. They all work perfect.

As I tried to combine your boards config file with some older ones I also got this NullPointer Exception. It looks like your config-file is not compatible with the one for ESP8266 boards. I get such a Null Pointer Exception, as soon as I use both of them at the same time and enter the Board Manager. Use these URLs to reproduce:
http://arduino.esp8266.com/package_esp8266com_index.json,https://raw.githubusercontent.com/mikaelpatel/Cosa/master/package_cosa_index.json

I had no problems combining Cosa with ATtiny or Digispark Boards.

I also use Mac OS X 10.10.3 and Arduino 1.6.4, so it may also be a platform specific problem. 

@dgschwend: Do you have some other Board-Config files included?

```
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$3.run(ContributionManagerUI.java:155)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributedToolReference.resolve(ContributedToolReference.java:43)
	at cc.arduino.contributions.packages.ContributedPlatform.resolveToolsDependencies(ContributedPlatform.java:76)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:92)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:600)
	at processing.app.Base$10.onIndexesUpdated(Base.java:1279)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$3.run(ContributionManagerUI.java:153)
	... 1 more
java.lang.NullPointerException
```

tht, 2015-06-04:
May be relevant: http://forum.arduino.cc/index.php?topic=324095.0

dgschwend, 2015-06-04:
Indeed, I have exactly the same ESP8266 board config file and it seems that's what's causing the problem. Installs and compiles fine without the ESP8266 config file. Awesome work @mikaelpatel!

mikaelpatel, 2015-06-04:
@tht @dgschwend 
Thanks for testing this update. It is a great improvement in handling the install of Cosa. Cheers!

tht, 2015-06-04:
I've forget your repository to try the same fix which worked for the attiny package mentioned above. And it does work here too. I can now use ATtiny, Digispark, ESP8266 and Cosa in one IDE - Perfect!

Thank you @mikaelpatel

tht, 2015-06-04:
Just did another test on a different computer. Works perfect now. I think we can close this issue…

closed 2015-06-04 by tht

rei-vilo, 2015-08-02:
That's really nice! 

Could you please promote the link https://raw.githubusercontent.com/mikaelpatel/Cosa/master/package_cosa_index.json on the front page so everyone can enjoy it?