mikaelpatel/Cosa #367: OutputPin::pulse(us) will have uint16_t overflow error if us > ((4*0xFFFF)/I_CPU)

dansut, 2015-06-04:
I think this is probably really just documentation - at least I can't think of an improvement to the code - but some sort of warning ought to exist that the number of microseconds can't use the whole uint16_t when the Frequency of the CPU is anything greater than 4MHz due to the way it is implemented.

Same doc need exists with OutputPin::write(val, bits, us), and maybe other places that use these...

Cheers
/dan

mikaelpatel, 2015-06-05:
@dansut 
I think this issue depends on the implementation of DELAY(us). https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/Types.h#L261
The calculation may overflow. There is yet another issue when using large bit delay values as the member function is synchronized (and locks out interrupts). https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/OutputPin.cpp#L46
The intended range is typically 1-100 us (for 8 bits). Both the range and that the function is synchronized (atomic) needs to be documented.

mikaelpatel, 2015-06-05:
@dansut 
BW there seems to a similar problem with synchronized in OutputPin::write(value, clk, order). https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/OutputPin.cpp#L24
The idea with a clock is to allow possible interrupts during the write().

mikaelpatel, 2015-06-08:
Please see update https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/OutputPin.hh#L281

dansut, 2015-06-08:
Looks like that should do the job Mikael, thanks.

I was going to suggest changing the us param to a uint8_t giving a range more inline with its intended usage and thus avoiding the hidden potential overflow due to implementation - but I guess this has potential for breaking existing code and warnings in the doc for this and the atomic issue should be more than adequate to keep users on the right track.

Cheers

closed 2015-06-08 by dansut