mikaelpatel/Cosa #371: Add support for LPC8XX

mikaelpatel, 2015-06-12:
NCP/ARM Cortex-M0+ MCU. http://www.nxp.com/news/press-releases/2012/11/nxp-revolutionizes-simplicity-with-lpc800.html

Possible easy start of adding support for ARM.

mikaelpatel, 2015-06-27:
Possible development board http://www.farnell.com/datasheets/1883897.pdf

tht, 2015-06-29:
+1

Would love to see Cosa on these little devices... If you're looking for a development board I can recommend this one: http://www.aliexpress.com/item/ARM-CortexM0-LPC812-Development-Board-Minimum-System-Board-USB-to-Serial-ISP-Download/32234512606.html
Small, cheap, embedded RGB LED for some testing and the integrated USB-UART Bridge helps a lot as lpc21isp is easily able to upload to these boards without any additional hardware.

jeditekunum, 2015-06-29:
+1 - maybe.

These would have limited value for me - only at the VERY lowest end. Good fit for AVR replacement. However, I wonder if the small size is going to be obsolete. When I can buy a Particle Photon with Cortex M3 + WiFi for $19...

closed 2015-09-20 by mikaelpatel