mikaelpatel/Cosa #376: Vote: Add support for esp8266 (WiFi module)

mikaelpatel, 2015-06-27:


HeMan, 2015-06-27:
Yes please!

SlashDevin, 2015-06-27:
As the dev tools for the ESP8266 module get better, combining it with an Arduino leverages the best of both worlds: lots of maker interest in the ESP8266 IoT, and lots of existing Arduino HW and SW.  Offloading the work usually associated with the W5xxx modules is very attractive, and allows the two "modules" to tackle what they are individually good at... it could be a very nice system partition.

So I think this gets my vote.  Although there is increasing interest in the ARM, there are *many* issues in supporting a new processor, as we have discussed.  ARM support has my second vote.  I suspect you can knock out this module fairly quickly, so maybe we'll get both?  We have unbridled confidence in your ability to single-handedly Cosa-fy the entire Arduino universe.   :)

mikaelpatel, 2015-06-27:
@HeMan @SlashDevin 
Thanks for all your support! 

@HeMan and I have previously discussed the ESP8266 module and there is actually two levels here. The first is using it as a WiFi module and the second is porting Cosa so that it may be used "within" the module and the GPIO pins. 

I found this LPC8XX based board http://www.farnell.com/datasheets/1883897.pdf and I think this is would a great starting point for porting to ARM. It is even pin compatible with Arduino Uno! Also Jeelabs is exploring the LPC8XX family. The mbed project is starting to become stable with regard to tooling so here is a great source for the build system. 

The latest refactoring of Cosa core is actually the ground work for a "HAL" with AVR and ARM. I am playing around with an operator syntax based variant of Cosa. This is mainly to see how far the operator syntax could be pushed. Pedagogically this is so much easier to get a grasp of. The Cosa OOP is a bit too "advanced" with all the multiple inheritance and templates classes.  And without any teaching material much too difficult to use as a beginner. 

Thanks again for the feedback!

jeditekunum, 2015-06-29:
It's been a few months since I last played with my ESP8266. It is VERY rough - unusable in my view.

After my struggle with the CC3000 (which I eventually abandoned) I don't have high expectations.

I'm waiting for my Particle Photon to arrive (July supposedly) for the project I need WiFi for. I don't plan to do any more projects with AVR.

rei-vilo, 2015-08-02:
I gave up ESP8266 after a while, mainly because of limited documentation.

For an Arduino-like WiFi board, I'm very happy with the [LaunchPad CC3200](http://embeddedcomputing.weebly.com/launchpad-wifi-cc3200.html) from Texas Instruments. 

The official IDE is [Energia](http://energy.nu), very similar to Arduino, except it provides 2 frameworks: 

* standard Wiring/Arduino framework, and 
* RTOS called [Energia Multi-Tasking](http://embeddedcomputing.weebly.com/exploring-rtos-with-galaxia.html).

closed 2015-09-20 by mikaelpatel