mikaelpatel/Cosa #377: Vote: Add support for Particle Photon (M3+WiFi)

jeditekunum, 2015-06-29:
http://docs.particle.io/photon/photon-datasheet/

Broadcom BCM43362 Wi-Fi chip
STM32F205 120Mhz ARM Cortex M3
1MB flash, 128KB RAM

$19 US.

mikaelpatel, 2015-06-29:
This is a serious board/chipset with resources for a true RTOS and full modular design. Need to have a look at the driver for the WiFi chip. Guess that the target is an embedded Linux kernel.

jeditekunum, 2015-06-30:
Linux would be painful on only 128KB memory. No doubt it will be supported by most RTOS soon.

This is the dilemma for Cosa expanding to ARM. ARM is now priced similar to AVR and is far more capable. The "competition" isn't going to be just the Arduino toolset - it's going to be every RTOS out there (FreeRTOS/Nuttx/RTEMS/...).

Board | Cortex | Flash | RAM | WiFi | US$
----- | ------ | ----- | --- | ---- | ---
LPC812 (https://github.com/mikaelpatel/Cosa/issues/371) | M0+ | 16KB | 4KB | no | $13
LPCXpresso824-MAX (https://github.com/mikaelpatel/Cosa/issues/371) | M0+ | 32KB | 8KB | no | $19
Teensy 3.1 (https://github.com/mikaelpatel/Cosa/issues/361) | M4 | 256KB | 64KB | no | $20
Photon | M3 | 1MB | 128KB | yes | $19

What we have here is **disruption**. LPC boards would have to be half the price in my mind. Generally (not counting clones from China) there are no AVR boards for less than ~$10 (Pro Mini 328 is ~$10 at SparkFun) so I think $10 is the minimum that manufacturers are going to bother with. I think we will see Cortex M3/M4 boards at $10 within a couple years - probably 512KB/128KB+.

So what does that mean for Cosa when the minimum hobbyist board is a "series board"?

mikaelpatel, 2015-06-30:
The plan forward is to continue the refactoring of Cosa is towards a core where AVR may be isolated with a HAL and extended to support ARM. This should work fine for the smaller ARM based MCU such as LPC824 but there is a sweet-spot where a RTOS or even embedded Linux kernel is the correct choice. 

I agree with you analysis of the "competition". The "end-game" is the amount of software and supported devices. Many free RTOS lack device drivers and the level of hardware and software understanding required is somewhat higher then Arduino libraries. Writing a device driver for embedded Linux is more "NHL" level programming. This is what gives a Cosa porting project some interest (I believe/guess) as it fills some of the gap between a simple core library as Arduino and a RTOS/commercial framework. 
 
A 16 MHz AVR (ATmega328P) can be more efficient than a 80 MHz ARM based MCU due to code and data footprint, and low-power. But with larger sketches and the need for more processing power the 80-160 MHz ARM with ATmega2560 amount of hardware resources takes over. 

Any framework/RTOS for the future will need firm software architecture to scale and be extended.

jeditekunum, 2015-06-30:
Device drivers are definitely a problem with the likes of FreeRTOS. Nuttx seems to be much better. Nuttx supports M0.

I'm inclined to think the low-power advantage of AVR is vanishing quickly. I just saw http://www.bluz.io/ which is an M0 board coming soon that has BLE and claims to be able to run "for years" on batteries.

What I'm trying to say is I think the niche that Cosa fills (even if ported to ARM) is being squeezed. The lean-and-mean nature of Cosa causes problems for more advanced uses. For example, network devices are notoriously asynchronous and, as we've seen with some devices, makes supporting them in Cosa too difficult.

I may not be the typical audience for Cosa (35+ years software engineering of "big iron" system software) but for me I need/want a more robust OS (even for hobby projects). My feeling is that I would use Nuttx on ARM going forward. Even if I can think of a project where a M0+ is appropriate I'd want to use the same base as I would for a M3, M4, and even R*/A*.

The amateur level audience is probably going to want at least the entire M series.

jeditekunum, 2015-07-03:
FYI Arduino form-factor for Photon: https://www.sparkfun.com/products/13321

rei-vilo, 2015-08-02:
For the ARM-based boards, the mbed SDK already provides RTOS capabilities. 

The real missing platform is Arduino.

closed 2015-09-20 by mikaelpatel