mikaelpatel/Cosa #386: Interrupt on Power Failure

jeditekunum, 2015-07-16:
Could AnalogComparator be enhanced to detect power failure?

mikaelpatel, 2015-07-17:
Sounds like an interesting idea as the BOD interrupt is a reset. How would the reference voltage be defined? The internal reference voltage could be used by AIN has to be a scaled from the power voltage (if I understand you correctly).

jeditekunum, 2015-07-17:
From what I understand, AIN1 would need to be supplied by an external voltage divider (driven off VCC) and AIN0 would be connected to the bandgap reference.

I believe this requires an enhancement to AnalogComparator to allow configuration of AIN0 to select the bandgap reference.

Not sure I want to use this approach due to additional external circuit. I'm currently using AnalogPin::bandgap in loop and am waiting for a small supercap to arrive. (I need to save a few bytes in EEPROM.) It may power things long enough that I don't need to use AnalogComparator interrupt...

mikaelpatel, 2015-07-17:
The simplest solution is to add a pseudo register (as AIN1) for the bandgap reference and enable it. I see that there is a problem with ADEN setting in the enable() member function. I will have a look into that as well.

mikaelpatel, 2015-07-19:
An alternative is to use the Brown-out detect and add a dispatch in the init() function (main.cpp). Something like "on_brown_out_reset()". This would be called before setup() which implies that the function will need to initiate any hardware needed to save state (for instance, some variables in SRAM to EEPROM). Also "on_power_on_reset()" could be introduced to allow reloading of state. 

All this could be automated by an iovec_t with pointers to SRAM variables and size, and a given EEPROM address.

Cheers! Mikael

jeditekunum, 2015-07-20:
My understanding of BOD is that low voltage triggers a BOD reset and that the processor is held in that reset until voltage rises above the threshold for some period of time (hysteresis). Further, enough voltage must be there during the whole process to maintain memory contents, etc. So it doesn't apply to complete power loss.

mikaelpatel, 2015-07-20:
@jeditekunum 
Hum, I should have read the title a bit better. Sorry about that. 

The BOD is not designed for this use case. The idea with a modification of the Analog Comparator seems to be a possible solution. I will do some testing and see what I come up with. Do you have a sketch of the circuit with the supercap? I guess the power supply line must go through a diode and the measurement done before it? AIN1 could be connected to a potentiometer together with VCC and GND. This would allow the low battery use case as well. 

Cheers! Mikael

jeditekunum, 2015-07-20:
I'm using a pro-mini clone - 328p.  Bandgap reference is 1.1v although can vary between 1 and 1.2. It seems to work fine with VCC of 3V (it is 3.3v version).

Diode shouldn't be necessary. I was thinking resister divider with VCC to AIN1 being about twice the value of AIN1 to GND. A pot would be good for testing - might need temporary capacitor over AIN1 and GND - not sure how sensitive comparator is.

Could also be on input side of regulator with appropriate change in resistor values. Diode probably needed then - and maybe zener clamp.

I don't have supercap yet (5.5V 0.047F on order). Not sure which side of regulator I will put it on. Supply side is regulated 5V from an inductive power circuit. Should be small enough that there won't be issue with charge current on either side of regulator.

mikaelpatel, 2015-07-25:
Please see commit https://github.com/mikaelpatel/Cosa/commit/4822efc3ac1ab4c71182b8b9b557e40ed9513afc.

jeditekunum, 2015-07-28:
Excellent! Thanks!

closed 2015-07-28 by jeditekunum

jeditekunum, 2015-08-03:
Just a followup with info relative to using a supercap. I connected a 5.5V 0.047F supercap at VCC pin of a pro-mini (3.3V version).

When I remove power it continues to operate for several seconds - plenty of time to save state. With this much time available the AnalogPin::bandgap reading works fine - no interrupt (or other external components) needed although that works fine as well.

mikaelpatel, 2015-08-03:
@jeditekunum 

Thanks for the followup. How did you end up connecting the "probe"? Did you need a diode?

jeditekunum, 2015-08-03:
@mikaelpatel 

To test I just put a 50K pot on the input power (which I'm supplying 5V to) and the wiper straight to the ADC pin.

To put this into a "production" use I suppose the "probe" would have to vary depending on a number of things. So perhaps a diode, resistor divider, and maybe a zener clamp.

As it turns out, the behavior of my project and the supercap means I won't be using this method for my current project.