mikaelpatel/Cosa #387: Vote: Cosa core with Arduino API

mikaelpatel, 2015-07-23:
Create a version of Cosa which uses the Arduino API.

rei-vilo, 2015-08-02:
If this simplifies the portability of the Cosa framework, let's try it!

jfpoilpret, 2015-08-03:
To be honest, I'm not sure I can see the interest here; Cosa brings 2 benefits:
- better performance
- good design, truly OO
What advantage would it be to use Arduino API? WOuldn't this remove at least one of the existing benefits?
Until I understand a good motivation for this, I would vote against.

mikaelpatel, 2015-08-03:
@jfpoilpret 

The idea is to refactor Cosa to a new "Arduino API" version to make it easier to use other Arduiono libraries and open a way to Cosa. This would be a totally different version of Cosa with its own repository.

jfpoilpret, 2015-08-03:
OK, I understand better now.

However, I am afraid this kind of stuff can take a lot of time and even though the benefits of using 3rd-party libs look appealing, I wonder about real feasibility, I mean I wonder if we would get assuanrance that ALL 3rd party libs could then work with Cosa without any change; I am afraid many libs would expose new issues in this Cosa Arduino API every day; but maybe I am wrong (I am a typical pessimistic person :-))

As per my own concern, I would neither vote for, nor against this proposal, as I don't see immediate benefits to my personal projects, but other people mileage may vary.

rei-vilo, 2015-08-03:
Some time ago, Hernando Barragan —the inventor of the Wiring framework— was working on a fully object-oriented framework called Wiring++ and even demoed it at Maker Faires. See this [post](http://feed.wiring.co/post/50831440404/wiring-try-the-demo-now).

[image]

mikaelpatel, 2015-08-03:
@rei-vilo 
I might as well put the cards on the table. Obviously this is not something I want to do but it could give a source code baseline that could be easier to compare with Arduino. One of my goals has been to inspire to higher quality code, both documentation and style of writing, for others to read. This does not seem to be one of the goals with the Arduino core. So far most discussion about architecture, quality, style and readability with the Arduino Team ends with them stating that they target beginners. 

Of all the proposed directions this is the least likely :)

rei-vilo, 2015-08-04:
Thank you for your interesting comments. There are now 2 Arduino teams: Arduino.CC and Arduino.ORG. Maybe you could try with Arduino.ORG. Wiring remains independent.

mikaelpatel, 2015-08-10:
Low priority and not a project goal.

closed 2015-08-10 by mikaelpatel