mikaelpatel/Cosa #391: Add support/wrapper for avr-crypto-lib

kc9jud, 2015-08-04:
Hello,

Upon finding my copy of Understanding Cryptography (great book!), I started wondering about cryptography on AVR, especially with regards to IoT... looking around, I found the [avr-crypto-lib](https://git.cryptolib.org/avr-crypto-lib.git) project... would anyone be interested in/have a use for a Cosa interface to this library?

Cheers,
Patrick

kc9jud, 2015-08-04:
Just to be clear, I'd be willing to work on this if there's sufficient interest.

mikaelpatel, 2015-08-04:
@kc9jud 
Thanks for the interesting idea and library reference. I had a quick look. It is not OOP/C++ and there are some assumptions on buffering etc that makes it difficult to wrap for an OOP framework. Also this is a very large library with lots of assembly which may give configuration problems. 

It is not a easy task to adapt a procedural function library to an OOP framework and OOP design patterns such as Decorator on IOStream (which is the most elegant for compression and cryto). 

If there is interest we could look into the class architecture for supporting crypto and then select and number of algorithms to refactor. This is obviously hard work but could be fun.

mikaelpatel, 2015-08-08:
@kc9jud 
Yes that is a good example of the need for an OOP framework to allow streaming of data through decorators into the Ethernet/W5100 buffer. The Cosa W5100 Ethernet Device driver is designed to not consume any extra buffers in SRAM instead the W5100 internal buffers are used. This is a very efficient design for an 8-bit MCU. The crypto library is not designed for this. It requires intermediate buffers which will consume extra memory that we basically dont have on the smaller AVR MCUs.

closed 2015-08-20 by mikaelpatel