mikaelpatel/Cosa #392: Brownout state?

jeditekunum, 2015-08-05:
Looking for ideas here.

I have a project that is inductive powered. I detect the loss of power to save state by reading the bandgap in loop. I have a small supercap on the regulated side of a pro mini. This provides plenty of time to detect the dropping voltage and save state. After I save state I call exit().

If I wait 15+ seconds before bringing back in range of the inductive power everything works fine - it restarts normally and my logic retrieves the saved state from eeprom. (BTW I write a "valid" pattern to eeprom right after saving state and I invalidate the pattern after reading the state - to ensure it isn't corrupt.)

If I don't wait long enough before reapplying power, its dead and stays that way - until removing power long enough or pushing reset.

The problem is that the supercap is draining slowly and its a question of whether it gets low enough to pass the point at which power up will trigger a POR (power on reset). If not, it should be a BOD reset. (BTW, my symptom is the same whether I have BOD disabled or not.)

Cosa currently has no handling of a BOD reset as far as I can tell. My understanding is that MCUSR has BORF set on reset if it was caused by BOD. I don't think handling is necessary as, in my case at least, I don't care why a reset happens - I'm just expecting a reset after power is reapplied.

So why isn't it resetting? I have BOD enabled. I tried it with cli() commented out in exit() to no effect.

Any ideas?

mikaelpatel, 2015-08-05:
@jeditekunum 
Does the MCU have a bootloader? If I remember correctly some bootloaders destroy the MCUSR contents.

jeditekunum, 2015-08-05:
@mikaelpatel

Its a pro mini clone with, I assume to be, a typical bootloader.

Even if it clobbered the MCUSR contents, would it matter? Cosa doesn't look at those flags.

Are you suggesting that maybe the bootloader won't start the app for a brownout reset? That might actually explain it.

jeditekunum, 2015-08-06:
It appears the normal bootloader for pro-mini doesn't do anything special - it does clear MCUSR.

I replaced my call to exit() with a loop printing out the value of the bandgap. Impressively, it goes for over 10 seconds and down to 1.9V before stopping. If I reapply power before that point the readings go back up - no brownout reset no matter whether BOD is disabled in Cosa or not. I don't understand this as the normal fuses are for 2.7V brownout.

On a related note, should Cosa have a facility for a software reset?

mikaelpatel, 2015-08-06:
@jeditekunum 
There is a sandbox sketch with a software reset. The problem is resetting the hardware registers.
https://github.com/mikaelpatel/Cosa/blob/master/examples/Sandbox/CosaRestart/CosaRestart.ino

The missing BOD reset is mystery; 1.9V is clearly below the 2.7V threshold :). I will have to rig a test of this.

jeditekunum, 2015-08-06:
@mikaelpatel 

Wouldn't using watchdog reset be better for software reset?

I imagine this would require some coordination with Watchdog::

mikaelpatel, 2015-08-06:
@jeditekunum 
Yes, that is the correct way to do that. And now we get to why I asked about the bootloader. As the register is cleared by the bootloader it is difficult to detect the software reset. Also the bootloader may use the watchdog to determine that the load is finished and call the application.

jeditekunum, 2015-08-06:
@mikaelpatel 

Maybe I'm naive about bootloaders but why do we care? Aren't the resets essentially the same except for leaving the "reason" flags set for the bootloader? Why should Cosa care what the reason was?

When I glanced at the pro mini bootloader I think it only checked for an external reset.

I imagine that the Watchdog vector in Cosa would have to be reset first to force default handling. Although I don't recall CDC doing anything special. In fact, CDC, minus the caterina magic word that means download, should be an example of how to implement a Cosa reset function.

jeditekunum, 2015-08-07:
@mikaelpatel 

This is odd. I tried another pro-mini clone from the same order and it does stop when it drops below 2.7V.

Still takes a fair bit of time to drop that much so I still have to deal with the period between having finished saving state and it actually stopping.

I guess I have to find a smallish 10000uF.

closed 2015-08-07 by jeditekunum

mikaelpatel, 2015-08-08:
@jeditekunum 
Thanks for all the info. This is great stuff you are working on. 

I had a discussion during the Spring with Rick at Anarduino. He was designing a similar set up (ie save to EEPROM/RTC RAM). The target was extreme low power where only a RTC was active during the "sleep period". A RTC signal is used to control the power feed to the MCU and wake it periodically to run sensors, log and transmit measurements. The state is saved in RTC RAM. 

There seems to be a state-store/restore pattern that is common for low power/power failure design that might be interesting to add to the Cosa core. The pattern is more or less a state save/restore to EEPROM/RTC RAM/etc and stop/start detect. 

A simple API for this would be a list of pointers and block sizes in program memory and a block address on the stable storage device. 

In any case, thanks again for the info and keeping me posted on your progress.