mikaelpatel/Cosa #394: Additional DHT11/22 device driver using InputCapture

mikaelpatel, 2015-08-10:
The InputCapture class gives an unique opportunity to implement a DHT11/22 device driver that is not as vulnerable to interrupts. 

The current DHT11/22 device driver uses ExternalInterruptPin and RTC micro-seconds. There is a high latency and the captures pulses may be corrupted by other ISRs. The InputCapture allows the timer value to be captured on the event and ISR latency more or less ignored.

guiguid, 2015-09-08:
I confirm using 2 DHT22 + serial I/O + I²C  is a pain.
DHT driver throw EXT0 and EXT1 is not stable.
Sometime communication is ready with one DHT, but rarely both, and often none !

mikaelpatel, 2015-09-08:
@guiguid 
Thanks for the feedback - appreciated. 

I have not done that type of test but it is as suspected. Unfortunately using Input Capture does not really solve the problem when using several DHTs as there is only one input pin (on the ATmega328). It would require for instance an analog mux. 

The current implementation more or less assumes that there are other interrupts after initiating a sensor read. The alternative is to "roll-back" the previous implementation with a synchronized block for the read cycle (worst case 5ms, 40 bit with 50+70 us). 

Meanwhile please reduce the interrupt load before reading the DHT sensor, e.g. flush serial data, etc. 

I guess that it is better to have a secure and stable sensor read than the possibility to handle interrupts during the read. The 1-wire protocol is design with each bit initiated by the host. This allows interleaving of interrupts. The DHT is difficult to sample due to the pulse width variation. Manchester code would have been so much easier.

closed 2015-09-20 by mikaelpatel