mikaelpatel/Cosa #413: Vote: Add teaching material

mikaelpatel, 2015-09-11:
There is a need for more information about the design behind Cosa and how to use all the classes. The doxygen source documentation and all the examples are great but for some teaching help could be appropriate. 

There are a few central themes in the architecture of Cosa that needs to be explained for those who do not read and study code. There are lots of historical notes that also could be explained with links to encourage readers to study further. 

There are also a number of design patterns that are used over and over again in the code to give decoupling and extension. One example is the interface style for device drivers with Northern and Southern bound interfaces.

jfpoilpret, 2015-09-12:
+1 
Cosa is fantastic but it's true it is quite hard to get into, essentially because it lacks (from my viewpoint) some kind of "overview doc" (main concepts and main classes) that would give some guidance **path** to learn it (ie: where should I start?) 

Maybe also some tutorial, to put those main concepts and classes into practice, would be useful too.

mikaelpatel, 2015-09-12:
@jfpoilpret 
Thanks for the feedback. Yes, I have seen that there is some difficulties getting started especially for those who have not used C++ or OOP before.  

So far I have seems a need for at least the following additional documentation:

1. Introduction to object-oriented programming.
2. The fundamentals of Cosa with the main concepts and classes (how to use perspective).
3. Introduction to design patterns and how they improve programs.
4. The main design patterns and architecture of Cosa (how it is constructed perspective).
5. How to add design drivers to Cosa.
6. Advanced Cosa programming; debugging techniques, architecture of applications, time and resources, battery based applications and low-power software design, etc. 
7. Application themes and frameworks.

I added this issue to try to understand who are the potential users, what is their typical back-ground in programming, and what would they need to better get started. 

There is a group of users that seem to get started very quickly and do very interesting projects. Most of them have done a lot of programming and have used C++. For them Cosa is very much what they have been looking for to program Arduino. 

Unfortunately there is also a group of users that more or less construct programs by "cut and paste". It is very difficult to help this group as them do not see the need to learn programming but just want to complete their project. Obviously this is how most of us get started before we understand that it is all about learning. 

@jfpoilpret 
Last thanks for "spreading the word" about Cosa. Much appreciated!

jfpoilpret, 2015-09-13:
For my personal situation, I already had good OOP and C++ knowledge when I started using Cosa, so it was not a barrier to using it.
For other people, maybe just links to OOP, C++ would be enough (newcomers would choose or not to go and read them).

I am not sure I understand your suggested point *7. Application themes and frameworks*, what would this contain?

Besides that, I fully second your list of topics in this order, which seems the most practical to follow for people new with Cosa.

mikaelpatel, 2015-09-13:
@jfpoilpret 
I do fullty agree with providing links to OOP and C++ introduction. There is no need to put the effort in that area. The issue is more what to assume that the user has understood of OOP and C++. Here I was thinking along the line of providing links from the source code. There is already a lot of info in the source code but I am not sure that user actually read source code.

The Application themes and frameworks has to do with higher level OOP structure. There are components that put together allow applications to be written as simple extensions.

gelius59, 2015-09-15:
Excellent project Cosa.
Create better documentation, a great idea.

closed 2015-10-16 by mikaelpatel