mikaelpatel/Cosa #425: Soft::UART receive no longer working?

jeditekunum, 2015-09-23:
It seems this is no longer working. I am using latest source. CosaSoftUART only changed to different pins (D25 & PCI0 on Moteino Mega). I have tried several other PCI pins.

Test output is fine but it doesn't echo back any input. My larger app isn't seeing input either.

mikaelpatel, 2015-09-23:
@jeditekunum 
Have you tried lower baudrates? I changed the PinChangeInterrupt implementation to save some SRAM. Tested on ATtiny. I will try it on ATmega and see.

jeditekunum, 2015-09-23:
@mikaelpatel 

Yes, 9600. At first with GPS. Then tried example with FTDI.

mikaelpatel, 2015-09-24:
@jeditekunum 
The CosaSoftUART example sketch works fine with Pro-Mini and the default setup D4(RX) and D5(TX) and 9600/57600 bps. I will have to test a few more configurations. Will get back with that.

mikaelpatel, 2015-09-24:
@jeditekunum 
Did some more tests with Mega. Worked as it should. Found an issue with ATmega328 (Standard) based boards and PCI14 and up. The numbering was wrong when using analog pins for PCI. https://github.com/mikaelpatel/Cosa/commit/7e0033f9892c9426c3c502848bcd8f4c5238be9d. Also the Mega PCI pin definition needed some cleaning up as some of the pins where based on analog pin definitions which makes it extra difficult to read. https://github.com/mikaelpatel/Cosa/commit/10199c560216907eaa8bbf574f5838ab838eb1ce

I will have to take a look at the Moteino Mega board definition again.

jeditekunum, 2015-09-24:
@mikaelpatel 

It is possible that this never worked with Moteino Mega as I never tested it when I did the board stuff :)

I hooked up my FTDI to uart1 to confirm that it is working. CosaMegaUARTs works fine.

I then moved to CosaSoftUART and moved my FTDI to the default pins. Nothing (not even output). Then changed baud rate to 9600; nothing. Changed to D0/D1(PCI9) and output worked but no input. Then changed to D1/D2(PCI10) and same thing.

Likely I did something wrong with the board stuff.

mikaelpatel, 2015-09-24:
@jeditekunum 
I had a look at the pin mapping for PinChangeInterrupt. It is confusing and needs to be cleaned up. The mapping has to be the same as the corresponding pin as it is sub-classes from IOPin. But at the same time it has to be a mapping to the PCI interrupt and mask. I will look at this.

jeditekunum, 2015-09-24:
@mikaelpatel 

Thanks. I used ATmega1284P.hh as the baseline so it will need cleanup too.

mikaelpatel, 2015-09-24:
@jeditekunum 
Actually the ATmega1248P.hh should be just fine as it is a straight mapping with ports in natural order. It is not restricted to a physical layout as a board. I guess that the Microduino Core++ has the same issue. The new implementation with a list of handlers instead of a full table reveals any mapping errors as the lists are per port/mask register. The table version did not really care if the index was "correct" as long as the mapping was the same for register and indexing.

mikaelpatel, 2015-09-24:
@jeditekunum 
There was several things to clean up in the PinChangeInterrupt code. Also the board definitions needed some work. When possible the PCIn are the same as the boards Dn. Special cases are MCU that have only a limited number of PCI-pins. Please see commit https://github.com/mikaelpatel/Cosa/commit/a13c5267fb90766b89706c88ebb2bdea82c2a28b.

jeditekunum, 2015-09-25:
@mikaelpatel 

Works! Thanks!

closed 2015-09-25 by jeditekunum

mikaelpatel, 2015-09-25:
@jeditekunum 
Thanks for the testing and feedback. The code was in need of a clean-up. Looked like I had left it in mid-air :). Must have got carried away with the UART optimization :)