mikaelpatel/Cosa #428: UART input garbage @ 115200 on atmega328p

jeditekunum, 2015-10-04:
```
#include "Cosa/IOStream/Driver/UART.hh"

void setup()
{
  uart.begin(115200);
}

void loop()
{
  while (uart.available())
    uart.putchar(uart.getchar());
}
```

Using miniterm, if you just type some stuff and hit return it will be echoed flawlessly. If you copy a block of text and paste it will be garbage. But if you change to 57600 the same paste works.

I switched to atmega1284p and it works fine even at 115200.

Pro Mini clone 5V/16Mhz, Adafruit FTDI Friend, OSX.

mikaelpatel, 2015-10-04:
@jeditekunum 
Have you tested changing the buffer sizes? With 32 character buffer the paste might overrun the buffer. I will have a look at miniterm so that it is using two stop bits.

jeditekunum, 2015-10-04:
@mikaelpatel 
Yep. 128 in and out. The garbage isn't just some characters - it looks like everything is bad.

mikaelpatel, 2015-10-04:
@jeditekunum 
Hum. Tested the snippet with standard buffers on nano and pro-mini without any issues. Strange?!? That leaves OSX as I am on Ubuntu. My guess it the stop bits.

mikaelpatel, 2015-10-04:
@jeditekunum 
Tested 230400 bps and got it to fail on pro-mini. Added stopbits=2 to miniterm and it worked nicely again. I will push an update.

jeditekunum, 2015-10-04:
@mikaelpatel 
Yes, stopbits 2 is working for me. Odd that miniterm doesn't have a way to set that on command line.

mikaelpatel, 2015-10-04:
@jeditekunum 
Nice catch! I have tested 230400 and pasting 8X80 character lines. Works nicely now.

jeditekunum, 2015-10-04:
@mikaelpatel 
Verified! Thanks!

closed 2015-10-04 by jeditekunum