mikaelpatel/Cosa #437: Arduino-Cosa on Atom Editor

neu-rah, 2015-10-17:
Hi, I've just published and integration of Arduino-Cosa with Atom Editor, it a very immature package as it only works on Linux platform (for now).

https://atom.io/packages/atomino

it works great with my breaduino (atmega328-8) and my USBASP ISP under Linux, but I'm sure a lot has to be done.
I'm open to suggestions.

mikaelpatel, 2015-10-17:
Nice work! I do not really know what to do with this issue as it is not a bug report or a feature request. Could you consider this announcement on the Arduino forum. There is a Cosa new topic there. I think that it the place if you wish some response. http://forum.arduino.cc/index.php?topic=150299.0

I would add a link if that is what you wanted with this issue?

neu-rah, 2015-10-17:
consider it an enhancement maybe and close it.
sorry for the communication channel not being the most adequate

mikaelpatel, 2015-10-17:
@neu-rah 
There are several alternative Arduino IDEs that include Cosa. I think I need to add a list of them. This would be the natural place for your editor. Cosa does not bundle any editor or IDE. It is a core with functionality for the boards that are support. Your editor is therefore not an enhancement by definition. It is good work! I would still encourage you to use the Arduino forum to attract helping hands. 

I use GNU Emacs and a make based build script (see build/cosa). This gives me great productivity, more battery time on my laptop and allows me to use all the processor cores in the build (:-). I do not really like IDEs such as Eclipse. It is just too much "whistles and flutes". I like to write code and not spend time on moving the mouse around and clicking. 

There are a number of human factors research reports that show that productivity drops when the context is switched too much. This includes moving between keyboard and mouse, moving from window to window, menu, etc. 

There are some things that I do actually like with modern IDEs. These have to do with navigating and analyzing code, keeping track of things that have changed, have to be done, etc. So I do use them when I have to do some "exploring". But at the end of the day they are just too slow for me. 

Are you planning any special functionality compared to other IDEs? What are your plans with this editor/IDE? Would you like to integrate with the Cosa Debugger support? That would be something extra compared to the Arduino IDE. What is your motivation?

neu-rah, 2015-10-17:
Integration is the main motif that dragged me here before going to arduino forum. Its pleasant to know that you have included debugger support, Have to explore that.
Until now I've just linked to Compile/Upload (ispload, need to review programmers), serial monitor, boards list, mem. size, board config, reset, clean.
The output is just dumped to a terminal window.
Some command-line options I've just got from a forum and I'm still exploring the documentation...
Reset has not worked for me, maybe something related to this ISP that I'm using. I have to explore some boards that I have around (arduino nano, arduino micro, attiny, atmega1284, atmega2560, and possibly others I can borrow from friends).
I was looking to the terminal dump and exploring documentation to support goto line error on mouse click.

Do you include any tools to detect ports and boards?
That would be a great enhancement. Right now I just rely on defaults or user settings to get things working.

My motivations was powered by arduino IDE and some frustrations with the poor architecture and some stuff that arduino IDE does in my back and that annoys me some times, being a C++ coder for a long time I just hate having code inserted and reordered in my back and sometimes specially when using templates, arduino IDE misses the point completely. Would prefer a more straight C/C++ environment.

Just curious, Is Cosa planning for support other uC beside AVR's? like ARM's, I have an arduino Due around.

Can Cosa work without requiring arduino IDE installation? can't see why not...

I'm a software guy and I don't do much of hardware stuff, I know about Cosa for quite a while and have followed your great work, but I have not used arduino or cosa that much.

Thanks for your enlightening response, I'm seeking Cosa debug docs right now.

mikaelpatel, 2015-10-21:
@neu-rah 
Some answers to your questions:

> Do you include any tools to detect ports and boards?

Cosa includes build support based on make and shell scripts. If you check the makefile (Arduino.mk) you will find some basic scripts for this. See https://github.com/mikaelpatel/Cosa/tree/master/build

> Just curious, Is Cosa planning for support other uC beside AVR's? like ARM's, I have an arduino Due around.

There is nothing stopping supporting other MCUs such as ARM more than time, resources and projects. Cosa is constructed to with classes, virtual member functions, variant handling, etc, so that this is possible. Cosa is not driven or financed by selling boards, consulting, or teaching. It is a spin-off from some of my home-automation projects. 

> Can Cosa work without requiring arduino IDE installation? can't see why not...

You can simply install the latest version of avr-gcc. There is no need to use the Arduino IDE or the AVR tools bundled there. See https://github.com/mikaelpatel/Cosa/blob/master/INSTALL.txt

If you want a powerful and small IDE and are using Linux (e.g. Ubuntu) I would suggest using Geany. Configured with the Cosa build script it is very easy to use. The only drawback is that there is no easy way to add a board menu. 

Using the Cosa build script is very easy. The following command compiles for uno, uploads and starts the serial monitor (miniterm.py).
```
cosa uno avanti
```
I use this from within Emacs. F2 to build, F3 to jump to error line or assert. You could use this for the build.

closed 2015-10-21 by mikaelpatel