mikaelpatel/Cosa #445: Add support for Arduino IDE 1.6.6 and new package manager

mikaelpatel, 2015-11-06:
The tool chain structure has also been changed. This has ripple effects on the Cosa command line build. The current work around is a number of symb-links.

mikaelpatel, 2015-11-16:
Initial support implemented. See commit https://github.com/mikaelpatel/Cosa/commit/2bc3c5bfaca3f06e548c1e831107719f72d1ef48.

traianpop, 2015-11-24:
I'm trying to run Cosa 1.1.1 installed through Arduino IDE 1.6.6 on a MacOSX (ElCapitan).

The sketch **Verify** operation could not find the compiler at first, but now it works fine after I defined `runtime.tools.avr-gcc.path` in `platform.txt`.

However, I need a suggestion on how to fix the **Upload** on an Arduino UNO. As far as I can see, the IDE messes up the `avrdude` path:

`___REMOVE___/bin/avrdude -C/Users/tp/Library/Arduino15/packages/Cosa/hardware/avr/1.1.1/build/avrdude.conf -v -v -v -v -patmega328p -carduino -P/dev/cu.usbmodem1A121 -b115200 -D -Uflash:w:/Users/tp/Documents/Arduino/build/cosablink.ino.hex:i
java.io.IOException: Cannot run program "___REMOVE___/bin/avrdude": error=2, No such file or directory
..." `

Obviously, the upload succeeds if I correct the above command and run it in a shell, but the upload is not working through the 1.6.6 IDE.

mikaelpatel, 2015-11-24:
I more or less only work on Linux (to save time :). Do some testing on Windows. Actually to get the latest version working on Linux I added some sym-links to the tools. The tools are handled as packages in 1.6.6. 

The __REMOVE__ is the interesting part of the path. Where is the IDE looking? 

The version 1.6.6 will soon be replaced/updated. There are too many issues with configuration. 

Why not use the Cosa command line build instead? This requires some configuration but after that you have ultra-fast parallel build with library caching etc. 

Cheers!

traianpop, 2015-11-24:
I don't know where the IDE is looking, I was hoping you had the answer. Otherwise, as you mentioned, there are plenty of alternatives to go around this issue. Plus I noticed from searching online that 1.6.6 has a lot of problems. I'll stick to command line.

closed 2015-11-26 by mikaelpatel