mikaelpatel/Cosa #446: Can't build any Cosa project :

guiguid, 2015-11-08:
the sketch (hard to make simpler): 

void setup() {
}
void loop() {
}

/usr/share/arduino/hardware/tools/avr/bin/avr-gcc-ar rcs /tmp/build8822524449374858952.tmp/core.a /tmp/build8822524449374858952.tmp/AnalogComparator.cpp.o 

sorry - this program has been built without plugin support

GNU ar (GNU Binutils) 2.25.1
avr-gcc (GCC) 5.2.0
Linux manjaro 4.2.5-1-MANJARO
Arduino 1.6.5

ar is compiled with "enable-plugins" .... 

I don't understand ....

If I chose a "non Cosa card type", it works

guiguid, 2015-11-08:
works perfectly with Cosa 1.0.3

mikaelpatel, 2015-11-08:
Can you check what ar is used. https://github.com/mikaelpatel/Cosa/blob/master/platform.txt#L46

mikaelpatel, 2015-11-08:
The answer was obvious if I checked the path of example. 

Try changing back to avr-ar.

guiguid, 2015-11-08:
it works with : 
compiler.ar.cmd=avr-ar
it fails with 
compiler.ar.cmd=avr-gcc-ar

mikaelpatel, 2015-11-11:
I guess this implies that we need to remove LTO support from the Arduino IDE build (again).

closed 2015-11-21 by mikaelpatel