mikaelpatel/Cosa #454: Add a Pin Change Interrupt Manager

joede, 2015-11-25:
As far as I could see your current pin change handler class allows handler based on the whole register. If the controller comes with i.e. 4 registers ('PCMSK_MAX' for 'PC1'..'PC4'), then up to four handlers can be registered.
The problem is, that in the case a library installs a handler for one pin on 'PC1', the whole port is "in use". Nevertheless, if the application starts using a free pin on the same port, the previous handler will be overwritten.
A better solution is a layer on top of the 'PinChangeInterrupt' class. Adding a manager class would allow a list of handlers for each port. This will add more flexibility at the cost of some CPU time. The user can decide whether to use a manager or the 'PinChangeInterrupt' directly.
This isn't my idea. I've found an implementation here at Github: https://github.com/prampec/arduino-pcimanager

mikaelpatel, 2015-11-25:
I am sorry to say that this does not sound like the Cosa PinChangeInterrupt https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/PinChangeInterrupt.cpp.

It uses a different strategy. The user level handler is per pin used (not port). Handlers are not overwritten if they do not use the same pin. They are already "managers". 

The dispatch is ISR(PCIn) => static on_interrupt(port...) => dynamic pin->on_interrupt().

I had a quick look at the link and the Cosa implementation does all that and more much more :) as it uses an abstract interrupt::handler interface. This makes it interchangeable with external interrupt pins when needed. I have not seen any other design or implementation that does this. 

Cheers!

joede, 2015-11-25:
After a short look, I've only seen `s_pin[Board::PCMSK_MAX]` and overlooked the comment with the "list" inside.
Just a second stupid question before closing the issue. ;-)
I'm browsing the source here in github and I can't find the code for the list. Where is the method `next()` of the class `PinChangeInterrupt` coming from? I couldn't find a parent class which implements it.

mikaelpatel, 2015-11-25:
@joede 
I had a further look at the link you provided. There is a lot of issues with that implementation especially as all listeners are in a single list! Also all listeners are called on an interrupt and each does a digitalRead. There is no masking and no change direction detect. I would avoid using that implementation. There is so much better code available :)

Guess you have seen the Cosa Soft UART implementation with PinChangeInterrupt. Compare with Arduino SoftwareSerial. That implementation is a source of many issues (as well) as it add PCI ISRs and overwrites any other implementation. This is the original issue: https://groups.google.com/a/arduino.cc/forum/#!topic/developers/CNEn5gqNWBo 

Last but not least, the Cosa PinChangeInterrupt is designed to handle all the supported boards from ATtiny to ATmega.

Cheers!

PS: Please feel free to close this issue.

joede, 2015-11-25:
@mikaelpatel I know that the referenced implementation isn't complete, it was just to link to the "ideas provider". ;-)
By the way... I have two questions which are no real issues. So I take the liberty to send you an e-mail. ;-)

closed 2015-11-25 by joede