mikaelpatel/Cosa #471: Driver: Improve W5X00 ethernet shield/device detect

mikaelpatel, 2016-01-07:


dansut, 2016-01-08:
I'm likely going to spend some time in the next couple of weeks following your lead with the work you did on the W5200 integration replicating for the W5500, as it is what I'm using on my project, so if you add any ideas you have or direction you'd like to take to this issue then I'll do anything I can to help.

Cheers
/dan

mikaelpatel, 2016-01-12:
@dansut 

After reading a number of forum topics on the Ethernet Shield and library it seems like there are some basic validations missing. 

1. Is a shield with the Ethernet controller available?
2. Is the Ethernet controller cable connected (link working)?

Both these should (if possible) be validated with the device driver is started. The first seems easiest to do by checking an SPI read as the return values are very specific (at least what I have seen for the W5100). The second issue is a bit trick as this is not available in a status register (what I know) and without a mod of the board it is difficult to "read the link status". Worst case this has to be ignored and higher level protocol has to detect that the network (link) is not available. Do you have any ideas on this?

Cheers! Mikael

dansut, 2016-01-15:
Good info, thanks - it strikes me using this type of technique it would perhaps be possible to automatically select the correct wiznet chip dynamically, although this is probably not so desirable with limited memory and no desire for binaries to contain stuff they will never use.