mikaelpatel/Cosa #474: Invalid Library

Testato, 2016-01-30:
Hi Mikael,

I installed COSA on IDE 1.6.7 by .json package.
When i choose board: Cosa Arduino Uno in the IDE consolle appear:
`Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\AVR: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\AVR
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\HTTP: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\HTTP
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ICMP: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ICMP
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\LCD: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\LCD
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\MQTT: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\MQTT
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\NEXA: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\NEXA
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\NTP: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\NTP
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ProtocolBuffer: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ProtocolBuffer
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Shell: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Shell
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\SNMP: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\SNMP
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Telnet: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Telnet
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ThingSpeak: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ThingSpeak
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\UML: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\UML
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Wireless: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Wireless
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\AVR: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\AVR
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\HTTP: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\HTTP
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ICMP: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ICMP
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\LCD: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\LCD
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\MQTT: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\MQTT
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\NEXA: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\NEXA
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\NTP: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\NTP
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ProtocolBuffer: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ProtocolBuffer
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Shell: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Shell
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\SNMP: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\SNMP
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Telnet: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Telnet
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ThingSpeak: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\ThingSpeak
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\UML: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\UML
Invalid library found in D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Wireless: D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\libraries\Wireless`

mikaelpatel, 2016-01-30:
This is basically just warnings that the Cosa internal libraries do not have library properties. It can be ignored as the Library Manager is not used for this. 

Library properties could be added but as they can only be used by the Cosa core there is no motivation other than a way to suppress these warnings from the Arduino IDE. There is still a lot of issues with the new go based Arduino IDE pre-processor.

mikaelpatel, 2016-01-30:
@Testato 
There is a note http://forum.arduino.cc/index.php?topic=150299.msg2529178#msg2529178 about issues with 1.6.7. This will not remove the warnings from the Library Manager.

mikaelpatel, 2016-01-30:
@Testato 
A more detailed look at the list of warnings show that they are for all the libraries that are git sub-modules. That is one of the issues with the Library Manager. It does not understand that. Please see the documentation about libraries https://github.com/mikaelpatel/Cosa/blob/master/doc/05-libraries.md.

Testato, 2016-01-30:
i do not know if it is related or not, but if i try to build the CosaBlinkClassic:

> Arduino: 1.6.7 (Windows 7), Board: "Cosa Arduino Uno (ATmega328/Optiboot)"

Warning: platform.txt from core 'Cosa AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
open preproc\ctags_target_for_gcc_minus_e.cpp: The system cannot find the file specified.
Error compiling.

Testato, 2016-01-30:
With verbose:

> Arduino: 1.6.7 (Windows 7), Board: "Cosa Arduino Uno (ATmega328/Optiboot)"
> 
> D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\arduino-builder -dump-prefs -logger=machine -hardware "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\hardware" -hardware "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages" -hardware "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware" -tools "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\tools-builder" -tools "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\hardware\tools\avr" -tools "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages" -built-in-libraries "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\libraries" -libraries "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\libraries" -fqbn=Cosa:avr:uno -ide-version=10607 -build-path "C:\Users\win7\AppData\Local\Temp\build6e8f3bf796d48e258821b84380c91822.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\examples\Blink\CosaBlinkClassic\CosaBlinkClassic.ino"
> D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\arduino-builder -compile -logger=machine -hardware "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\hardware" -hardware "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages" -hardware "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware" -tools "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\tools-builder" -tools "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\hardware\tools\avr" -tools "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages" -built-in-libraries "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\libraries" -libraries "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\libraries" -fqbn=Cosa:avr:uno -ide-version=10607 -build-path "C:\Users\win7\AppData\Local\Temp\build6e8f3bf796d48e258821b84380c91822.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\examples\Blink\CosaBlinkClassic\CosaBlinkClassic.ino"
> Warning: platform.txt from core 'Cosa AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
> "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\hardware\tools\avr/bin/avr-g++"  -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -Woverloaded-virtual -Wextra -std=gnu++11 -felide-constructors -fno-implement-inlines -fno-rtti -fno-threadsafe-statics -mcall-prologues "-IC:\Users\win7\AppData\Local\Temp\build6e8f3bf796d48e258821b84380c91822.tmp"       "-ID:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\cores\cosa" "-ID:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\variants\arduino\uno" "C:\Users\win7\AppData\Local\Temp\build6e8f3bf796d48e258821b84380c91822.tmp\sketch\CosaBlinkClassic.ino.cpp"
> "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\hardware\tools\avr/bin/avr-g++"  -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -Woverloaded-virtual -Wextra -std=gnu++11 -felide-constructors -fno-implement-inlines -fno-rtti -fno-threadsafe-statics -mcall-prologues "-IC:\Users\win7\AppData\Local\Temp\build6e8f3bf796d48e258821b84380c91822.tmp"       "-ID:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\cores\cosa" "-ID:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\variants\arduino\uno" "C:\Users\win7\AppData\Local\Temp\build6e8f3bf796d48e258821b84380c91822.tmp\sketch\CosaBlinkClassic.ino.cpp"
> "D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\hardware\tools\avr/bin/avr-g++"  -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -Woverloaded-virtual -Wextra -std=gnu++11 -felide-constructors -fno-implement-inlines -fno-rtti -fno-threadsafe-statics -mcall-prologues "-IC:\Users\win7\AppData\Local\Temp\build6e8f3bf796d48e258821b84380c91822.tmp"       "-ID:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\cores\cosa" "-ID:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\arduino-1.6.7\portable\packages\Cosa\hardware\avr\1.1.1\variants\arduino\uno" "C:\Users\win7\AppData\Local\Temp\build6e8f3bf796d48e258821b84380c91822.tmp\sketch\CosaBlinkClassic.ino.cpp"
> open C:\Users\win7\AppData\Local\Temp\build6e8f3bf796d48e258821b84380c91822.tmp\preproc\ctags_target_for_gcc_minus_e.cpp: The system cannot find the file specified.
> Error compiling.

mikaelpatel, 2016-01-30:
@Testato 
The release note stated that Cosa can not be installed via the Boards Manager in 1.6.7. There is a release pending. Instead please install via the ZIP master. Alternatively copy the latest platform.txt file.

Testato, 2016-01-30:
Yep, 
with the last platform.txt file there is no error
Thanks

closed 2016-01-30 by Testato

mikaelpatel, 2016-01-30:
@Testato 
New release available with support for install with the Boards Manager. https://github.com/mikaelpatel/Cosa/releases/tag/v1.2.0a

mikaelpatel, 2016-01-31:
@Testato 
The sub-module library warning should be removed now. Please reinstall R1.2.0. https://github.com/mikaelpatel/Cosa/releases/tag/v1.2.0b

Testato, 2016-01-31:
Yes, tested.

No warning whit the last Board Manager update

Testato, 2016-01-31:
Yes, tested.
No warning
I installed it via Board Manager on an Official Portable IDE version

mikaelpatel, 2016-01-31:
Great. Thanks for your effort testing. Appreciated.