mikaelpatel/Cosa #475: Submodules are not included in the Boards Manager install file

per1234, 2016-01-31:
The folders that are supposed to contain the submodules are empty. This causes the warning:
```
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\AVR: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\AVR
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\HTTP: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\HTTP
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\ICMP: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\ICMP
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\LCD: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\LCD
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\MQTT: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\MQTT
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\NEXA: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\NEXA
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\NTP: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\NTP
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\ProtocolBuffer: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\ProtocolBuffer
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\Shell: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\Shell
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\SNMP: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\SNMP
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\Telnet: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\Telnet
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\ThingSpeak: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\ThingSpeak
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\UML: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\UML
Invalid library found in C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\Wireless: C:\Users\per\AppData\Local\Arduino15\packages\Cosa\hardware\avr\1.2.0\libraries\Wireless
```

per1234, 2016-01-31:
Just notice this was already reported in https://github.com/mikaelpatel/Cosa/issues/474, sorry for duplicate issue.

closed 2016-01-31 by per1234

mikaelpatel, 2016-01-31:
@per1234 
This issue should be removed in the update of R1.2.0. Please reinstall; https://github.com/mikaelpatel/Cosa/releases/tag/v1.2.0b