mikaelpatel/Cosa #482: using rotary encoder connected to I/O expander

nitinsane, 2016-03-21:
I have a rotary encoder connected to the two digital pins of an I/O expander. Is there any way to use Cosa code to support this? I do see the DigitalPin abstraction but the IO Expander classes do not use it. Any ideas?

mikaelpatel, 2016-03-22:
What type of I/O expander? If you are thinking about using an I2C variant I think that the possible interrupt/sample rate is much too slow.

nitinsane, 2016-03-22:
I am planning to use MCP23017/8. I have not yet done any experimentation to see what kind of response I get.

kc9jud, 2016-03-23:
I might suggest using the a chip dedicated to reading out rotary encoders -- I've used http://www.lsicsi.com/counters.htm before and been very happy. I have a Cosa-compatible library I wrote for the LS7366R that I could clean up and contribute if anyone has a use for it.

nitinsane, 2016-03-23:
That's an interesting idea. Please share the code if you can.
Thanks.

closed 2016-04-08 by mikaelpatel