mikaelpatel/Cosa #484: PlatformIO support

yuroller, 2016-06-01:
I would like very much support for Cosa framework inside the build system [PlatformIO](http://platformio.org/).
By patching platformio with https://gist.github.com/SinishaDjukic/5e059f56b58fb9db3483 (see also my comment to cosa.py) and copying Cosa framework, I am able to build code in windows for arduino uno, but it would be nice if everything is more integrated (i.e. download, support for other boards).
What do you think?

mikaelpatel, 2016-06-01:
Have you tried the command line build that is already available in Cosa? I have previously been approached by the author of PlatformIO. At that time I came to the conclusion that there was no added value, just a lot of work to supply all the library descriptions, etc. Cosa is an integrated environment for medium to large scale high end projects with nearly 200 classes. There are only a few second source libraries. Adding yet another build environment takes time from producing industry strength code :) It is a question about priority. The estimated man-hours to add all the necessary files (NB no added Cosa functionality) is in the range 40-80 h including testing and packaging. 

PlatformIO does not automatically give support for other boards. That is a different ball game. 

Last but least Cosa is currently open-source and non-commercial, totally funded by myself. At the current usage, feedback, and spin-off rate this project will unfortunately close by the end of this year. 

Cheers!

yuroller, 2016-06-01:
I think PlatformIO would give more visibility to Cosa framework and make application development easier: just a few command lines and with platformio you can download build tools, framework and be productive in minutes.
I tried to build Cosa on windows, but it is not so easy, unless you have the full arduino ide installed (the makefiles have explicit reference to arduino paths), but then you are limited to arduino IDE; with PlatformIO you can choose between many development IDEs.
You can point the gist url in my previous post to the PlatformIO developer that approached you, so he can give an estimate on the effort needed to have a basic support for Cosa into PlatformIO.
I hope you will continue developing Cosa, as it is much more advanced than the default Arduino framework.

mikaelpatel, 2016-11-05:
See commit https://github.com/mikaelpatel/Cosa/commit/c654f78e4aaa7b9e349fb5476886f83534ac53ff.

closed 2016-11-05 by mikaelpatel