mikaelpatel/Cosa #489: Installation on Ubuntu without arduino

Dzenik, 2016-10-18:
Follow the installation instructions INSTALL.txt faced with the following problems:

> 3. Set environment variable GCC_AVR_VER to gcc-4.2.
   GCC_AVR_VER=gcc-avr-4.2 ; export GCC_AVR_VER

After installation apt-get install gcc-avr  I did not find anything gcc-avr, there is only avr-gcc-4.8.2.
Ok, I set GCC_AVR_VER=avr-gcc-4.8.2 and export it

> 5. Run command line build with board.
   cd Cosa/example/Benchmarks/CosaBenchmarksPins
   cosa nano avanti

I receive error: avr-g++: error: missing device or architecture after '-mmcu='
After viewing the scripts, I find that there is no boards.txt.
Ok, downloading  arduino archive, unpack and copy boards.txt in ~/Cosa/build/
Аgain I am trying to compile.
Got error: 
/home/evgen/Cosa/cores/cosa/Cosa/Board.hh:31:19: fatal error: Board.h: No such file or directory

The question is, Cosa does not work without Arduino IDE package?

mikaelpatel, 2016-10-18:
I see that the INSTALL.txt is out of date and the Arduino.mk needs updating. New version available. 

Please install: 
`sudo apt-get install gcc-avr binutils-avr gdb-avr avr-libc avrdude`

And copy boards.txt from the root directory.
```
cd Cosa
cp boards.txt build
```
To verify.
```
cd Cosa/build
GCC_AVR_VER=gcc-4.9.2 ; export GCC_AVR_VER
cosa uno
```
Cheers!

Dzenik, 2016-10-18:
Thanks, compiled success

closed 2016-10-18 by Dzenik