mikaelpatel/Cosa #496: API Documentation link is stale...

WestfW, 2017-04-11:
The link in the readme file refers to a non-existent file...
http://dl.dropbox.com/u/993383/Cosa/doc/html/index.html

mikaelpatel, 2017-04-11:
Seems like dropbox has changed their policy for handling public folders. Need to rethink how to store on-line documentation.

WestfW, 2017-04-13:
I noticed recently that github displays markdown and acts on links therein for files in a repository (not just in their "wiki" sections.)   So if you're not getting complicated with your html, that might be an alternative.  I don't know how hard the conversion would be.

gregori, 2017-08-31:
Is it possible to provide the documentation somewhere else for download?
As for publishing online, what about using github pages?

mikaelpatel, 2017-08-31:
@WestfW @gregori 
Time to clean up this mess. Sorry for the delay. 
https://mikaelpatel.github.io/Cosa/index.html
The documentation zip may be found in the release handling on github. 
https://github.com/mikaelpatel/Cosa/releases

closed 2017-08-31 by mikaelpatel