mikaelpatel/Cosa #503: board .map file could use better namespace allocation

liamkinne, 2017-12-31:
in every map file the arrays look like this:
```c++
const Board::DigitalPin digital_pin_map[] __PROGMEM = {
  Board::D0,
  Board::D1,
  Board::D2,
  Board::D3
}
```
couldn't it look like this?
```c++
using namespace Board;
const DigitalPin digital_pin_map[] __PROGMEM = {
  D0,
  D1,
  D2,
  D3
}
```
Doesn't change much, just saves space and improves readability.

mikaelpatel, 2018-01-01:
@liamkinne Did you try this? I am curious as Board is declared as a class and not a namespace. My guess is that you would get a compile error. In any case the ripple change is making all the Boards namespaces and then changing all the mapping files. 

Actually I like to enforce full naming as a good practice. In the long run it makes code more easy to maintain.

liamkinne, 2018-01-01:
Right, I was assuming it was a name-space. It makes a lot more sense now I know it's a class.

I'd assume there is some way of making this cleaner. I'll have a look.

closed 2018-06-30 by mikaelpatel