mikaelpatel/Cosa #507: TWI::Slave broken

dothebart, 2018-05-22:
As discussed in #305 the slave example is currently broken. 
At the time #305 was closed, the slave example was working

Two seeduino lotus provisioned with the slave and the master example from that revision can talk to each other. 
The regression that broke the slave again was introduced with https://github.com/mikaelpatel/Cosa/commit/173c3ebaa483d09acefb62df920bc6b064bf4420 - older slaves could talk to newer servers; newer couldn't. 

Submitting this as a new issue since #305 was fixed.

mikaelpatel, 2018-05-23:
Yes, the update to the TWI event handling has broken the TWI::Slave implementation. Busy right now but should get time to check the details in a few weeks. Thanks for the reporting.

dhillondeep, 2018-06-05:
@Mogball

Mogball, 2018-06-05:
Yeah we aren't to receive events either

dhillondeep, 2018-06-29:
@shahzadlone tried to fix the commit [173c3eb](https://github.com/mikaelpatel/Cosa/commit/173c3ebaa483d09acefb62df920bc6b064bf4420) and he made changes committed here [on his branch](https://github.com/shahzadlone/Cosa/commit/29a83a29d50cc439447b9a1c3c7001aba62a6c70).
```cpp
void
TWI::isr_stop(State state, uint8_t type)
{
  TWCR = TWI::STOP_CMD;
  loop_until_bit_is_clear(TWCR, TWSTO);
  if (UNLIKELY(state == TWI::ERROR_STATE)) m_count = -1;
  m_state = state;

  // Check for asynchronous mode and call completion callback
  if (m_dev->is_async() || m_status == SR_STOP) {
    m_dev->on_completion(type, m_count);
    // m_dev = NULL;
    m_busy = false;
    // TWCR = 0;
  }
}
```
This seems to fix the issue and the slave starts to work but, slave breaks again on the commit a0d6795c04086423fb25fbd232c56a3e7ce9caa2 where support for low power mode was improved. Can you suggest what we should look into for a potential fix?

mikaelpatel, 2018-06-29:
After reviewing the changes after the low power support the issue seems to be the missing powerup() in TWI::Slave::begin.

shahzadlone, 2018-06-30:
You are right @mikaelpatel that seemed to have fixed it. I will be putting all these bug fixes in and then open a pull request.

mikaelpatel, 2018-06-30:
Please see commit https://github.com/mikaelpatel/Cosa/commit/2b07bd6ac18b4350030c764e8ae2a938b323ac42.

closed 2018-06-30 by mikaelpatel

dothebart, 2018-07-02:
btw, is there a list of opensource projects using cosa? Probably @dhillondeep could add an item to it? ;-)

dhillondeep, 2018-07-04:
@dothebart I do not think there is such a list. So I ended up creating a list and you can add to it https://github.com/dhillondeep/cosa-projects

Mogball, 2018-07-04:
What about our sensor modules? :P not just our build tools

dhillondeep, 2018-07-04:
@Mogball you can add those modules if you want rn. I will add some when I have more time

shahzadlone, 2018-07-08:
This should now be fixed.