mikaelpatel/Cosa #510: [RFM 69] High-level API for Encryption Key and Frequency Selection

veecue, 2018-08-04:
Hi,
the RFM69 API seems pretty basic and does not support important features like setting an encryption key or selecting a frequency out of the box. Could that be fixed?

mikaelpatel, 2018-08-05:
@veecue I can only agree with you on this issue. The RFM69 driver does not expose all the available functionality, it only implements the Wireless::Driver API, which is very basic. There are several ways to fix this (or extend the functionality). One, which I would recommend, is to add member functions to the RFM69 API and implement them only for this driver (i.e. do not first update the Wireless::Driver API). 

I guess we are talk about basically two new member functions for setting an encryption key and setting a frequency. All the configuration registers are defined so the new functions should be straight forward (inspiration can be found in https://github.com/LowPowerLab/RFM69/blob/master/RFM69.cpp#L422). 

Cheers! 

And thanks for you interest in Cosa.

veecue, 2018-08-05:
Ok, so I already would go your recommended way. I'm currently working at a project using the RFM69 and wanted to implement it with Cosa (which is really awesome btw), so I'll just be extending the RFM69 class as I go (encryption, freq, ack, what else is needed) and send a PR when I'm finished.

closed 2018-10-15 by mikaelpatel