mikaelpatel/Cosa #511: Update toolchain version in JSON file to IDE 1.8.9

joede, 2019-04-09:
The current tarball 1.8.9 of the Arduino IDE comes with the following tool versions:

~~~~
arduino.avrdude=6.3.0-arduino14
arduino.arduinoOTA=1.2.1
arduino.avr-gcc=5.4.0-atmel3.6.1-arduino2
~~~~

Cosa 1.2.0 specifies avr-gcc *4.8.1-arduino5* and avrdude *6.0.1-arduino5*!

To avoid duplicate toolchain installations, could you please update the board manager file (JSON) or duplicate it with a "1.8.x" variant?

Are there any reasons not to update these versions?

joede, 2019-04-09:
For some reason, I have started arduino from the command line. I have already added your package manager JSON file to the IDE.

The steer shows the following messages:

Es konnte keine boards.txt in /home/jd/work/Arduino/sketchbook/hardware/Cosa/boards gefunden werden. Ist sie noch von einer Version vor Arduino 1.5?
Es konnte keine boards.txt in /home/jd/work/Arduino/sketchbook/hardware/Cosa/libraries gefunden werden. Ist sie noch von einer Version vor Arduino 1.5?
Es konnte keine boards.txt in /home/jd/work/Arduino/sketchbook/hardware/Cosa/examples gefunden werden. Ist sie noch von einer Version vor Arduino 1.5?
Es konnte keine boards.txt in /home/jd/work/Arduino/sketchbook/hardware/Cosa/build gefunden werden. Ist sie noch von einer Version vor Arduino 1.5?
Es konnte keine boards.txt in /home/jd/work/Arduino/sketchbook/hardware/Cosa/variants gefunden werden. Ist sie noch von einer Version vor Arduino 1.5?
Es konnte keine boards.txt in /home/jd/work/Arduino/sketchbook/hardware/Cosa/config gefunden werden. Ist sie noch von einer Version vor Arduino 1.5?
Es konnte keine boards.txt in /home/jd/work/Arduino/sketchbook/hardware/Cosa/bootloaders gefunden werden. Ist sie noch von einer Version vor Arduino 1.5?
Es konnte keine boards.txt in /home/jd/work/Arduino/sketchbook/hardware/Cosa/cores gefunden werden. Ist sie noch von einer Version vor Arduino 1.5?
WARNING: Error loading hardware folder /home/jd/work/Arduino/sketchbook/hardware/Cosa
  Im Ordner Cosa wurden keine gültigen Hardwaredefinitionen gefunden.

mikaelpatel, 2019-04-10:
> Are there any reasons not to update these versions?

Time and interest :) 

There has been limited interest in this project so it is on ice. I have actually moved my focus to developing a strategy for "porting" Cosa to support multiple architectures e.g. SAM and AVR. The Arduino-GPIO/TWI/SPI/OWI repositories are examples of the experimentation. 

I will consider the update. Thanks for the attention!